inkcut/inkcut #13: Feature requests

frmdstryr, 2017-12-07:
Please submit any feature requests you would like to see in the new version.

Thanks!

matthijskooijman, 2017-12-15:
See also https://github.com/slandis/InkCutter/issues

frmdstryr, 2017-12-15:
I believe all of those are fixed in this release.

The only thing that is not done is automatic text conversion (rects and groups are now supported). It maybe will be added in the future.

[image]

frmdstryr, 2017-12-19:
A request was made to have different translations.

I am also adding Python 3 and Qt5 support (as it is no longer using gtk).

raboof, 2018-03-14:
Maybe we should close this one and ask people to submit separate wishlist issues?

closed 2018-03-14 by frmdstryr