inkcut/inkcut #14: New logo / desktop icon

frmdstryr, 2017-12-15:
If anyone would like to contribute a new logo / desktop icon to use please feel let me know.  If there's multiple I can open up a vote on which to use.

It will be used for the desktop icon.  This is the current logo

[image]

srdco, 2018-04-18:
Did you create a new logo?

frmdstryr, 2018-04-18:
Never got around to it, feel free to make one! Would be great to have something a bit more modern like inkscapes logo.

RRREEEEEEEE, 2019-02-20:
I really like the logo, are you still looking for something else? I can ask my company to give it some thought, we have 7 artists.

frmdstryr, 2019-02-21:
I think the current logo is a bit dated. I'm not much of a designer :+1: so a professional design would be great!

RRREEEEEEEE, 2019-02-22:
hahaha they all like it, one of the girls is going to try to take out shadows and flatten it, but we'll see what they come up with I said to keep it in mind and I'll post here whatever they produce.

G2G2G2G, 2019-05-11:
They haven't done much but this was just an idea one has been messing with since February. 
https://i.imgur.com/ABnPL6m.png

drewrwilson, 2019-06-06:
Here's a quick one that I made to use locally as a desktop icon. Feel free to use if you'd like (or edit in any way)! 

[image]

Download SVG: [inkcut-icon.svg.zip](https://github.com/inkcut/inkcut/files/3262855/inkcut-icon.svg.zip)