inkcut/inkcut #22: Importing Text nodes from Inkscape

raboof, 2018-02-28:
I would really like to be able to seamlessly send documents with Text nodes from Inkscape to InkCut.

The 'proper' way to do this, of course, would be to add text-to-path support in InkCut itself: that would work for any .svg, whether it comes from Inkscape or not. However, this could get complicated with respect to finding installed fonts, 'rendering' etc.

A quick 'hack' might be to do a 'behind the scenes' Inkscape conversion. I've done this for the 'plot' extension (https://gitlab.com/inkscape/inkscape/merge_requests/163, https://gitlab.com/inkscape/inkscape/merge_requests/201) and while technically somewhat terrible, results in a nice non-destructive workflow.

Would you be interested in a PR with the latter solution, or are you confident the first could work and be willing to provide some guidance there?

frmdstryr, 2018-03-01:
Yes this would be great to have, but I expected it to be a lot of work so I left it out as it wasn't something that was absolutely needed.

The QPainterPath that is the core piece of inkcut supports adding text (see https://doc.qt.io/qt-5/qpainterpath.html#addText) and I believe it handles the "rendering" (via toSubpathPolygons) but I don't think it has all the features that inkscape has (such as individual spacing, etc). If inkscape creates individual nodes for each text element or something then it might be easy to do.  

I think a plugin that does the latter solution would be fine,  however I don't think requiring inkscape would be a good idea as I currently need to support the raspberry pi.  If it's something that can be turned on/off i'd be ok with pulling it in.

I'll leave it up to you as to which you want to do. Currently this is not something I need so i'll let you handle it and can help with any questions.

Thanks!

frmdstryr, 2018-03-01:
Also there would be no problems doing both (as I think any reimplementation will always create slightly different results).

enyachoke, 2020-05-08:
@frmdstryr  what the situation  it seems like it was merged then reverted. This is something I could like to have in my workflow and willing to put some time into. My preference could be to not require inscape at all.

raboof, 2020-05-08:
@enyachoke what part seems to have been reverted? #34 is still there on master - though it indeed does require inkscape.

enyachoke, 2020-05-08:
Saw this https://github.com/inkcut/inkcut/pull/35 thought it was the same but just noticed it was different

raboof, 2020-05-08:
See #36 :)