inkcut/inkcut #25: Send cut to device on parallel port

raboof, 2018-03-04:
For sending to a device that is connected to a parallel port, should that be done by somehow configuring the device as a 'printer', or would it make sense to add a third 'Parallel' option to the current 'Printer/Serial' list?

(the 2 cutters I have access to I can send files to by just sending them to `/dev/usb/lpX`)

frmdstryr, 2018-03-04:
I think most devices now have usb serial. 

The easiest way to do this would be to create a printer that goes to the parallel port then use that. I haven't tried and don't have the hardware to test this.  

If that doesn't work, adding a separate option for parallel wouldn't be hard to do if you're familiar with twisted. I can provide more info if you want to try to implement it.

raboof, 2018-03-14:
Added documentation on how to add a parallel cutter as a printer in #30, that should be sufficient

raboof, 2019-06-07:
Might be helpful/more discoverable to support this 'directly' after all, https://github.com/inkcut/inkcut/issues/168#issuecomment-499596700

frmdstryr, 2019-06-07:
Agree. Should be straightforward on linux, unsure about osx & windows.  

Do you know if there's a standard command in linux to find all lp ports? Or do you think just searching for

```
/dev/parport*
/dev/lp*
/dev/usb/lp*
```

Is sufficient? It'd be nice to automatically obtain info about each like pyserial does. I think pyusb would work for this when a usb adapter  is used.