inkcut/inkcut #31: Printer cannot reconnect

raboof, 2018-03-13:
I saw the following trace:

```
Traceback (most recent call last):
  File "/home/aengelen/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 150, in maybeDeferred
    result = f(*args, **kw)
  File "/home/aengelen/dev/inkcut/inkcut/device/transports/printer/plugin.py", line 200, in connect
    self.connection.transport = self
AttributeError: 'NoneType' object has no attribute 'transport'
```

It looks like the connection is intially populated:
```
    #: Delegate to the implementation based on the current platform
    connection = Instance(PrinterConnection)
```
Then on connection its transport is set:
```
    def connect(self):
        try:
            #: Save a reference
            self.protocol.transport = self
            self.connection.transport = self
            self.connection.open()
```
But then on disconnect the field is cleared:
```
    def disconnect(self):
        if self.connection:
            self.connection.close()
            log.debug("{} | closed".format(self.config.printer))
            self.connection = None
```
... causing the next connect to fail.

I guess the connection field should be repopulated? What would be the right place way to do that?

frmdstryr, 2018-03-13:
Ahh, good find. Can you try the above PR?  I've only been using the serial interface so the printer side is largely untested.

raboof, 2018-03-14:
Fixed in #32

closed 2018-03-14 by raboof