inkcut/inkcut #33: 'print to file'

raboof, 2018-03-13:
I have a plotter that I know works with some dialect of GPGL, but doesn't work with the GPGL produced by inkcut.

It'd be useful to be able to dump the data produced by inkcut. What would be the sensible place to plug in? As a third `Connection`, perhaps configurable with the filename?

frmdstryr, 2018-03-14:
Yes a new connection type for this would be the correct way to do it. 

Either I can do it, or if you'd like to try.  You must:

1. Create a new package here https://github.com/codelv/inkcut/tree/master/inkcut/device/transports using either of those as a template. The [TestTransport](https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L72) can be used as a reference if needed.
2. Then register it with the workbench here https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L898

If you're interested, there's more docs on the workbench here http://enaml.readthedocs.io/en/latest/dev_guides/workbenches.html

raboof, 2018-03-14:
Sounds good! Feel free to take it - I'll comment here if/when I start on it myself so we don't duplicate work.

frmdstryr, 2018-03-14:
Please try out https://github.com/codelv/inkcut/pull/38/commits when you have a chance. Thanks!

raboof, 2018-03-14:
Looks good! A small follow-up in #41 ;)

closed 2018-03-30 by raboof