inkcut/inkcut #37: Support for arcs

raboof, 2018-03-14:
inkcut doesn't appear to render arcs correctly, such as:
```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
 >
  <g
     id="layer1">
    <path
       id="path14"
       d="M 100.96745,167.77426 A 44.858372,157.69438 0 0 1 56.109067,325.46861 44.858372,157.69438 0 0 1 11.2507,167.77426 44.858372,157.69438 0 0 1 56.109067,10.079877 44.858372,157.69438 0 0 1 100.96745,167.77426 Z"
       style="opacity:0.93000034;fill:#ffffff;stroke:#000000;stroke-width:0.51863015;stroke-linejoin:round" />
  </g>
</svg>
```
which afaict should be an ellipse. Found http://pyqt.sourceforge.net/Docs/PyQt4/qpainterpath.html#arcTo-2 and https://www.w3.org/TR/SVG/paths.html#PathDataEllipticalArcCommands, which indeed don't seem to correspond 1-to-1

frmdstryr, 2018-03-14:
Interesting. Looks like the arguments need transformed.  

It would be good to get a bunch of different svg's together and build a test suite for this.

frmdstryr, 2018-03-15:
Looks like this should work https://github.com/qt/qtsvg/blob/5.11/src/svg/qsvghandler.cpp#L1482-L1546

raboof, 2018-03-16:
Nice find I'll give that a go

raboof, 2018-03-16:
Well, this was trickier than I expected ;). Putting this aside for now. (I did find https://www.w3.org/TR/SVG/implnote.html#ArcConversionEndpointToCenter as well, but haven't connected all the dots yet)

raboof, 2018-03-24:
I tried converting to an `arcTo` command (instead of to multiple `cubicTo` commands like the code you linked to), but I misunderstood the `arcTo` parameters: the x/y/w/h parameters describe the dimensions of the full ellipse, which means this API does not support arcs with a value for non-0 `x-axis-rotation`.

frmdstryr, 2018-03-24:
Just remembered the `cubicsuperpath.py` file in inkscape's extensions.  Looked and it has this

```python

def ArcToPath(p1,params):
    A=p1[:]
    rx,ry,teta,longflag,sweepflag,x2,y2=params[:]
    teta = teta*pi/180.0
    B=[x2,y2]
    if rx==0 or ry==0 or A==B:
        return([[A[:],A[:],A[:]],[B[:],B[:],B[:]]])
    mat=matprod((rotmat(teta),[[1/rx,0],[0,1/ry]],rotmat(-teta)))
    applymat(mat, A)
    applymat(mat, B)
    k=[-(B[1]-A[1]),B[0]-A[0]]
    d=k[0]*k[0]+k[1]*k[1]
    k[0]/=sqrt(d)
    k[1]/=sqrt(d)
    d=sqrt(max(0,1-d/4))
    if longflag==sweepflag:
        d*=-1
    O=[(B[0]+A[0])/2+d*k[0],(B[1]+A[1])/2+d*k[1]]
    OA=[A[0]-O[0],A[1]-O[1]]
    OB=[B[0]-O[0],B[1]-O[1]]
    start=acos(OA[0]/norm(OA))
    if OA[1]<0:
        start*=-1
    end=acos(OB[0]/norm(OB))
    if OB[1]<0:
        end*=-1

    if sweepflag and start>end:
        end +=2*pi
    if (not sweepflag) and start<end:
        end -=2*pi

    NbSectors=int(abs(start-end)*2/pi)+1
    dTeta=(end-start)/NbSectors
    #v=dTeta*2/pi*0.552
    #v=dTeta*2/pi*4*(sqrt(2)-1)/3
    v = 4*tan(dTeta/4)/3
    #if not sweepflag:
    #    v*=-1
    p=[]
    for i in range(0,NbSectors+1,1):
        angle=start+i*dTeta
        v1=[O[0]+cos(angle)-(-v)*sin(angle),O[1]+sin(angle)+(-v)*cos(angle)]
        pt=[O[0]+cos(angle)                ,O[1]+sin(angle)                ]
        v2=[O[0]+cos(angle)-  v *sin(angle),O[1]+sin(angle)+  v *cos(angle)]
        p.append([v1,pt,v2])
    p[ 0][0]=p[ 0][1][:]
    p[-1][2]=p[-1][1][:]

    mat=matprod((rotmat(teta),[[rx,0],[0,ry]],rotmat(-teta)))
    for pts in p:
        applymat(mat, pts[0])
        applymat(mat, pts[1])
        applymat(mat, pts[2])
    return(p)

```

frmdstryr, 2018-03-30:
Looking good, thank you!

I'm going to wait for #45 to before merging.

raboof, 2018-03-30:
I'm not really planning on implementing #45 right now - so I'd appreciate merging this, unless of course you're planning on doing #45 soon :).

frmdstryr, 2018-03-30:
Ok. One last request before merging, can you have it raise a `NotImplementedError` instead of just doing a move?

raboof, 2018-03-30:
Sure! Wouldn't it be nicer to show a warning but continue with just the supported elements? But perhaps that is for another PR

raboof, 2018-03-30:
Done ;)

raboof, 2018-04-06:
Let's close this since the common functionality was merged in #42 and missing advanced features are tracked in #45 and #46

closed 2018-04-06 by raboof