inkcut/inkcut #48: Sometimes plots with 'y' coordinate always 0?

raboof, 2018-03-30:
I did a print job where all the y coordinates somehow were zero:

```
2018-03-30 15:44:49,303 | INFO    | inkcut | device | Initializing job
2018-03-30 15:44:49,303 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x7f2815dcf290>
2018-03-30 15:44:49,303 | DEBUG   | inkcut | device | Path length: 2149.12377946
2018-03-30 15:44:54,382 | INFO    | inkcut | device | Connecting to device
2018-03-30 15:44:54,383 | INFO    | inkcut | device | Processing job
2018-03-30 15:44:54,383 | DEBUG   | inkcut | device | connect
2018-03-30 15:44:54,391 | DEBUG   | inkcut | -> smartlab | IN;
2018-03-30 15:44:54,392 | DEBUG   | inkcut | smartlab | opened
2018-03-30 15:44:54,392 | INFO    | inkcut | device | Working...
2018-03-30 15:44:54,394 | DEBUG   | inkcut | -> smartlab | PU2042,0;
2018-03-30 15:44:54,394 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:44:55,767 | DEBUG   | inkcut | -> smartlab | PD2037,0;
2018-03-30 15:44:55,768 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:44:55,792 | DEBUG   | inkcut | -> smartlab | PD2023,0;
2018-03-30 15:44:55,793 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:44:55,817 | DEBUG   | inkcut | -> smartlab | PD2001,0;
2018-03-30 15:44:55,818 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:44:55,842 | DEBUG   | inkcut | -> smartlab | PD1970,0;
2018-03-30 15:44:55,843 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:44:55,866 | DEBUG   | inkcut | -> smartlab | PD1932,0;
```

After opening and closing the settings dialog (but without changing anything AFAIR) the next print worked fine:

```
2018-03-30 15:45:35,550 | DEBUG   | inkcut | device | init <inkcut.job.models.Job object at 0x7f2815dcf290>
2018-03-30 15:45:35,550 | DEBUG   | inkcut | device | Path length: 2302.00373711
2018-03-30 15:45:37,168 | INFO    | inkcut | device | Connecting to device
2018-03-30 15:45:37,168 | INFO    | inkcut | device | Processing job
2018-03-30 15:45:37,169 | DEBUG   | inkcut | device | connect
2018-03-30 15:45:37,176 | DEBUG   | inkcut | -> smartlab | IN;
2018-03-30 15:45:37,177 | DEBUG   | inkcut | smartlab | opened
2018-03-30 15:45:37,177 | INFO    | inkcut | device | Working...
2018-03-30 15:45:37,180 | DEBUG   | inkcut | -> smartlab | PU2042,3403;
2018-03-30 15:45:37,180 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:45:38,154 | DEBUG   | inkcut | -> smartlab | PD2037,3496;
2018-03-30 15:45:38,155 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:45:38,179 | DEBUG   | inkcut | -> smartlab | PD2023,3586;
2018-03-30 15:45:38,180 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:45:38,203 | DEBUG   | inkcut | -> smartlab | PD2001,3673;
2018-03-30 15:45:38,204 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:45:38,225 | DEBUG   | inkcut | -> smartlab | PD1970,3756;
2018-03-30 15:45:38,226 | DEBUG   | inkcut | Rate is :0.36
2018-03-30 15:45:38,248 | DEBUG   | inkcut | -> smartlab | PD1932,3835;
```

frmdstryr, 2018-03-31:
Someone on gitter reported seeing this so thanks for creating an issue.

The only thing I can think of is that somewhere the model is getting mirrored twice when re-cutting and then this https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L503  is clipping it to 0. A change there from `min` to `abs` would work but ideally finding why it's being flipped twice would be best.  

I know it's flipped here https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L777 but it "should" always be creating a copy and leaving the original as is.  Originially [QPainterPath.__ mul __](http://pyqt.sourceforge.net/Docs/PyQt4/qpainterpath.html#__mul__-2) returned a copy which is why I implemented [imul](https://github.com/codelv/inkcut/blob/master/inkcut/core/svg.py#L52). Maybe that changed in PyQt5 (the PyQt5 docs don't have QPainterPath anymore)? 

The PreviewPlugin also applies this transform https://github.com/codelv/inkcut/blob/master/inkcut/preview/plugin.py#L98 but again that should always be creating a copy.

raboof, 2018-04-08:
I see it on python2 as well, it seems to always alternate between 'collapsed' and 'normal' when sending the same file to the device multiple times. Will look closer some time this week unless someone else beats me to it :)

raboof, 2018-04-09:
OK I think I figured out what's going on: because https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L463
```
        #: Move the job to the new origin
        x, y, z = self.origin
        model.translate(x, y)
```
happens before the model is flipped, but the https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L743:
```
            #: Set the origin
            if job.feed_to_end and job.info.status == 'complete':
                self.origin = self.position
```
happens after the model is flipped, the 'y' vector of the origin is reversed the second time around, causing the image to be translated from <0 to >0, then flipped to <0, and then disappears due to the 'max'.

So flipping the sign or the ordering of these operation 'fixes' the issue, but even after the fix the behavior seems kind of surprising... do we still want this 'origin' feature, especially since we already have the 'n copies' option?

frmdstryr, 2018-04-09:
Thanks for looking into this. Interestingly if you change it to

```python
# Move the job to the new origin
x, y, z = self.origin
model.translate(x, -y)
```
Then it works as expected. I'm not sure why this changed. 

The idea behind the `origin` was that eventually I wanted to be able to have a job queue so you could schedule several different plots at once and come back when they're all done (assuming the media doesn't need to change and the tracking is good enough to run several jobs).

raboof, 2018-04-10:
Aah I didn't consider a job queue with different plots, that is interesting. Though OTOH wouldn't you create one big SVG laying out all your elements, so you can arrange them optimally?

frmdstryr, 2018-04-10:
It depends on the workload, ours happens to use a lot of copies of the same graphic so the fill row feature takes care of this (it doesn't matter if we cut extra copies as they just go into inventory).

If the workload is multiple unique jobs, however, then yes each would need to be positioned individually.  This case would however ideally be done by a solver like https://github.com/nucleic/kiwi (the one enaml uses) to efficiently automatically do layout for you. 

Now thinking about it, I should have implemented this when I ported the old code over as it works in all cases (just would be changing the constraints).

raboof, 2018-04-10:
What do you think, should we remove this 'origin' field for now and leave more advanced workflows for future features?

frmdstryr, 2018-04-10:
I think adding the `-y` and maybe an option to disable it would be best.  It depends if the user/device resets the origin on every plot or not.

commit 96232591 (2018-04-10): Merge pull request #77 from codelv/fixFeedToEnd

Fix 'feed to end', disable it by default (fixes #48)
commit aea0674c (2018-04-10): Fix 'feed to end', disable it by default (fixes #48)
closed 2018-04-10 by frmdstryr