inkcut/inkcut #51: setup.py install: could not find enamlx

raboof, 2018-04-03:
When I try to `sudo python ./setup.py install`, the output ends with:

```
Installing inkcut script to /usr/local/bin
 
Installed /usr/local/lib/python2.7/dist-packages/inkcut-2.0.4-py2.7.egg
 
Installed /usr/local/lib/python2.7/dist-packages/inkcut-2.0.4-py2.7.egg
Processing dependencies for inkcut==2.0.4
Processing dependencies for inkcut==2.0.4
Searching for enamlx
Searching for enamlx
Reading https://pypi.python.org/simple/enamlx/
Reading https://pypi.python.org/simple/enamlx/
No local packages or working download links found for enamlx
error: Could not find suitable distribution for Requirement.parse('enamlx')
```

.. which is weird because 'pip install enamlx' worked fine.

(full logs at https://pastebin.com/Axnvmj5F)

srdco, 2018-04-03:
I have problems with that each time I installed it (Ubuntu 17.04 and UbuntuGnome 16.04).

I overcame it the last time by installing pip3 and using that to install the dependencies. I forget what I did the first time.

There seem to be 2 potential problems:

1) when both python2.7 and python3 are present (as is default in current Ubuntu), and pip install a python3 version, and then setup.py defaults to python2.7, or vice versa, or some other quirk of pip (which I am not terribly familiar with).

2) enamlx seems to ALWAYS have to be installed directly from github (I just added the relevant commands to the install-from-source docs a few days ago, which I believe have been merged in).

Hope this helps.

raboof, 2018-04-03:
Both 'pip' and 'python' are python 2.7 in this case:
```
➜  inkcut git:(master) ✗ python --version
Python 2.7.14
➜  inkcut git:(master) ✗ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
➜  inkcut git:(master) ✗
```
And indeed I can run inkcut fine, but I cannot install it through setup.py...

frmdstryr, 2019-03-22:
If anyone runs into this error use `pip install enamlx`

closed 2019-03-22 by frmdstryr