inkcut/inkcut #57: SVG scale?

raboof, 2018-04-03:
I created a ~26x26mm SVG in inkscape, but when loaded into inkcut it becomes 0.29x0.29 inch - which is too small. Do you have any pointers on how this is supposed to work?

[1x1cm-blocks.svg.txt](https://github.com/codelv/inkcut/files/1873302/1x1cm-blocks.svg.txt)

frmdstryr, 2018-04-05:
It seems a few things in Inkscape changed since I initially wrote Inkcut. Looks like inkscape is not using 90 ppi by default but 96 ppi.  

From  http://wiki.inkscape.org/wiki/index.php/Units_In_Inkscape

```
This is a nominal 10x6 inch drawing.

    The actual size of the drawing when displayed depends on how it is displayed. It is very unlikely to be the nominal size when displayed on a digital screen.
    The width of '10in' could be equivalently written as '25.4cm', '254mm', etc.
    The width could also be written as '960' or '960px' using the CSS/SVG defined value of 96 pixels per inch.
        If a 'unit identifier' is missing or 'px', the length is interpreted to be in 'user units'. CSS requires the use of a 'unit identifier'. Thus while one can write 'font-size="20"' (an SVG presentation attribute), one must write 'style="font-size:20px" (a CSS style value).
        Inkscape 0.91 and earlier used a value of 90 pixels per inch.
```

Also,  Inkcut is not transforming the root svg node to handle document unit changes. Ie, this needs implemented: https://github.com/codelv/inkcut/blob/master/inkcut/core/svg.py#L680-L685

raboof, 2018-04-05:
I'm learning more about SVG than I expected ;)

If I'm reading http://tutorials.jenkov.com/svg/svg-viewport-view-box.html correctly, this means we should set `self.viewBox` in the `QtSvgDoc` `__init__` and then take it into account in `parseUnit` when `isinstance(value, (int, float))`, right?

commit 90c7b004 (2018-04-07): Scale based on the viewBox (fixes #57)

Handles the common case that the viewBox is at the origin.
Also translating is left as separate issue #69
closed 2018-04-07 by frmdstryr