inkcut/inkcut #72: error installing

HAFEZZZ, 2018-04-07:
when i use (source venv activate )
i get ( bash: source: venv: is a directory

and when i use pip install . 

at the end i get 

(
Exception:
Traceback (most recent call last):
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/wheel.py", line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/home/plotter/.local/lib/python2.7/site-packages/pip/wheel.py", line 323, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/usr/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/zope.interface-4.4.3-py2.7-nspkg.pth'

frmdstryr, 2018-04-07:
It should be "source venv/bin/activate"

HAFEZZZ, 2018-04-07:
$ source venv activate
bash: source: venv: is a directory

ubuntu mate

HAFEZZZ, 2018-04-07:
i am using ubuntu mate and i did all the commands from the file directory 

plotter@plotter-OptiPlex-GX620:~/Desktop/plotter files/main/inkcut-master$ source venv activate
bash: source: venv: is a directory

HAFEZZZ, 2018-04-07:
even when i download the released version and extract to inkscape it opens the old version extension i cant install the new version with the scaling

frmdstryr, 2018-04-07:
You can also try just doing `pip3 install --user inkcut-master`

HAFEZZZ, 2018-04-07:
sorry i will

closed 2018-04-07 by HAFEZZZ