inkcut/inkcut #73: scaling

HAFEZZZ, 2018-04-07:
on the inkscape extension when i try to cut a 3cm square it cuts 3.2

and a 5cm is 5.33 

without margin 

i am using the cutter to cut a printed material i even try to calibrate the measures but without effect it works only with a simple square but with a complicated design still a calibration problem

frmdstryr, 2018-04-07:
You can change the scale for your device by:

1. Opening device setup from the "Device -> Setup" menu.
2. Clicking the Device tab
3. Clicking the output tab 
4. Setting the x,y scale to the value needed

HAFEZZZ, 2018-04-07:
i cant find the new inkscape extension 

the released version when i unpack it in the extension folders gives me the old extension

closed 2018-04-07 by HAFEZZZ