inkcut/inkcut #76: 'blade offset' support

raboof, 2018-04-09:


frmdstryr, 2018-04-09:
The original algorithm was a major hack.  

```python

	# ------------- plot export ---------------------------------
def applyOffset(self,hpgl):
    d = self.offset*self.scale
    def angleBetween(axis,p0,p1):
        def dotP(p0,p1):
            p = 0
            for a1,a2 in zip(p0,p1):
                p +=a1*a2
            return p                
        def norm(p0):
            n = 0
            for a in p0:
                n +=a*a
            return math.sqrt(n)    
        p0 = [p0[0]-axis[0],p0[1]-axis[1]]
        p1 = [p1[0]-axis[0],p1[1]-axis[1]]
        assert norm(p0) > 0 and norm(p1) > 0, "invalid points"
        r = dotP(p0,p1)/(norm(p0)*norm(p1))
        if -1 <= r <= 1:
                return math.acos(r)
        else:
            return math.pi
    
    def arcto(radius,theta,(x,y),p0):
        poly = []
        arc = ['A',[radius,radius,theta,0,1,x,y]]
        d = simplepath.formatPath([['M',p0],arc])
        p = cubicsuperpath.parsePath(d)
        cspsubdiv.cspsubdiv(p, self.smoothness*self.scale)
        for sp in p:
            first = True
            for csp in sp:
                if first:
                    first = False
                else:
                    for subpath in csp:    
                        poly.append(list(subpath))
        return poly
    def curveto(p0,curve,flat):
        poly = []
        d = simplepath.formatPath([['M',p0],curve])
        p = cubicsuperpath.parsePath(d)
        cspsubdiv.cspsubdiv(p, flat)
        for sp in p:
            first = True
            for csp in sp:
                if first:
                    first = False
                else:
                    for subpath in csp:    
                        poly.append(['L',list(subpath)])
        return poly
    
    
    
    if  d <= 0:
        return hpgl
    
    #poly = hpgl.parse(hpgl)
    
    last = ['PU',[0,0]] # start position 
    cur = [hpgl[0][:2],map(int,hpgl[0][2:].split(','))]
    i = 1
    while i < len(hpgl):
        cmd = hpgl[i][:2]
        point = map(int,hpgl[i][2:].split(','))
        next = [cmd,point]
        if not last[1] == cur[1] and not cur[1] == next[1]:
            theta = angleBetween(cur[1],last[1],next[1]) 
            if theta < math.pi/1.1:
                # start point
                dist = bezmisc.pointdistance(last[1],cur[1])
                t = d/dist+1
                start = bezmisc.tpoint(last[1],cur[1],t)
                # end point
                dist = bezmisc.pointdistance(cur[1],next[1])
                t = (4*d)/dist
                end = bezmisc.tpoint(cur[1],next[1],t)
                if t<=1:
                    # go distance d past actual point 
                    hpgl[i-1]='%s%d,%d'%(cur[0],start[0],start[1])
                
                    # curve to end point 
                    #for pt in arcto(d,0,end,start):
                    #    hpgl.insert(i,'%s%d,%d'%(cur[0],pt[0],pt[1]))
                    #    i+=1    
                    
                    # end point, probably dont need
                    hpgl.insert(i,'%s%d,%d'%(next[0],end[0],end[1]))
                    i+=1
        """
        # skip double points
        if not last[1] == cur[1] and not cur[1] == next[1]: # where are doubles coming from?
            # get the angle between the two vectors using cur as origin
            theta = angleBetween(cur[1],last[1],next[1]) 
            if theta < math.pi/1.1:
                # go past by offset amount
                dist = bezmisc.pointdistance(last[1],cur[1])
                t1 = d/dist+1
                start = bezmisc.tpoint(last[1],cur[1],t1)
                
                # come back to next line
                dist = bezmisc.pointdistance(cur[1],next[1])
                t2 = (4*d)/dist
                finish = bezmisc.tpoint(cur[1],next[1],t2)
                if t2 <= 1:
                    # add cur point
                    hpgl.insert(i-1,['L',list(start)])
                    
                    hpgl[i-1] = ['L',list(finish)]
                
                
                
        #else:
        #    inkex.debug("failed on %i with points %s,%s and %s"%(i,last,cur,next))    
        
        """
        # shift to next point
        last = cur[:]
        cur = next[:]
        i+=1
    return hpgl

```

It simply looked at the angle between two segments after it was broken into a polygon and modified the point to "draw past" by the offset amount if the angle was  "sharper" than an arbitrary value I found by guess and check.

I was intending to add a "filter" extension type that would allow plugins to be run on the path and apply these kinds of transforms (blade offset, overcut, whatever else...). I'm not sure if that's entirely necessary and probably overkill. What are your thoughts?

raboof, 2018-04-10:
I haven't encountered a printer that needs this feature, and I'm not sure how common they are - just noticed this was a feature that was in the old inkcut and not yet in this version.

I'd be completely fine with leaving this open at least until we find some evidence anyone actually cares :).

frmdstryr, 2018-04-11:
Pretty much all the cheap chinese cutters need this.

JesusFreke, 2018-10-20:
I thought this sort of functionality was required, to get accurate cuts on a cutter? Or are you saying that most cutters handle this in their firmware?

JesusFreke, 2018-10-21:
I just realized that inkscape's hpgl export has blade offset support now, so that works well for me.

Here's a quick comparison, if anyone's interested. The top was cut with inkcut v1.0, with it's... suboptimal offset algorithm :) And the bottom was with inkscape's hpgl export.

https://photos.app.goo.gl/oaEu1TFfAsYkrQqW8

The bottom looks a bit more crisp, while the top has some issues with the lines not quite matching up on the squares. e.g. that one square on the vertical trace in the upper left.

frmdstryr, 2018-10-21:
> are you saying that most cutters handle this in their firmware?

Yes, good cutters will do this in firmware.

Thanks for the comparison!  Interesting to see.

JesusFreke, 2018-10-21:
Gotcha, interesting. I only have experience with this cheap one, which obviously doesn't.

lazyzero, 2018-12-14:
I am also interested in such kind of feature.

frmdstryr, 2018-12-14:
Please give #122  a try. I don't have a device that needs this but it appears to look correct.

JesusFreke, 2018-12-15:
fwiw, inkscape's default hpgl export thing has offset support

frmdstryr, 2018-12-21:
I merged this. It uses a cleaned up version of the same algorithm. Please reopen or open a new issue if any are found.

closed 2018-12-21 by frmdstryr