inkcut/inkcut #78: Release

raboof, 2018-04-10:
With #77, I think we have everything we need to release this rewrite of inkcut feature-wise.

Is there anything missing? I guess there's work to do in documentation and building distribution packages / uploading to PyPI?

I guess we should do some 'beta' releases first, or do you think we should first call for some more feedback from people who are not afraid to install from source?

frmdstryr, 2018-04-10:
Uploading to pypi would be fine.

frmdstryr, 2018-04-17:
Perhaps https://github.com/astraw/stdeb would work?

raboof, 2018-04-17:
I agree it would be great to have deb packages - there seem to be a number of ways to achieve that (https://wiki.debian.org/Python/LibraryStyleGuide?action=show&redirect=Python%2FPackaging, https://github.com/paylogic/py2deb/blob/master/docs/comparisons.rst) and it might not be trivial - perhaps that should be a separate ticket?

Should we simply tag and upload 2.0.4 now? Or should it be 2.0.5?

srdco, 2018-04-17:
Have you looked into snaps?
https://www.ubuntu.com/desktop/snappy

frmdstryr, 2018-04-17:
Bump it to 2.0.5 as some fixes were added in. py2deb looks fine too.

I have used snaps before but I think a deb is still needed,  if you want to make a snap for it that'd be great!

raboof, 2018-04-25:
`python setup.py sdist; bdist_wheel --universal; twine upload --repository-url https://test.pypi.org/legacy/ dist/*` uploaded to test.pypi OK, but then `pip3 install inkcut==2.0.4` complained `faulthandler` had become builtin on python 3? Would appreciate if you could help/document :smiley:

frmdstryr, 2018-04-25:
Yes faulthandler is a builtin in py3.

raboof, 2018-04-25:
It certainly is, but I didn't expect that error message since setup.py correctly seemed to have make faulthandler conditional. Or perhaps it had to do with the fact that I built/uploaded with python2 and tried to use it with python3?

frmdstryr, 2018-04-25:
Yeah I actually tried the same thing.  I did one for each as it seems to just use the dependencies from the setup python version used to build it.

https://pypi.org/project/inkcut/

frmdstryr, 2018-04-25:
Does pypi have a way to create a "group" or "team" so multiple people can upload?

raboof, 2018-04-25:
Just signed up as 'raboof'

frmdstryr, 2018-04-25:
Thanks, added you so you can push updates.

raboof, 2018-05-26:
2.0.6 has been released to pypi

closed 2018-05-26 by raboof