inkcut/inkcut #86: Adding defaults for new devices?

srdco, 2018-04-18:
Was poking into the way to add 'automatic' defaults for new devices. I found the instructions to add a 'factory' but wasn't quite sure what was meant, other than the general idea of Python class factories. This is probably because I do not normally work in Python, but was wondering if there was a way to define standard defaults for an existing device (e.g. connection speed, return to origin set to true, x-axis mirrored set to true, etc) by a simple config file?

frmdstryr, 2018-04-18:
See https://github.com/codelv/inkcut/issues/56

frmdstryr, 2018-04-18:
Actually that doesnt let you specify default connection parameters. It might be good to add a similar way to that.

The idea behind the factory was it lets you define a function that returns a custom device subclass (or a preconfigured device instance).

srdco, 2018-04-18:
That would work. Does there happen to be an example of a preconfigured device instance? I poked at the VinylCutter RPI one, but didn't see something as directly applicable as I wanted. However, if I knew I was actually looking for something, I can poke a little more and maybe create one for the Vision Engraver. Particularly, I'm wanting to quickly be able to configure it no matter which computer I install Inkcut on without remember all the various settings required to make engraving work with this.

frmdstryr, 2018-04-18:
Something like this should work for the connection parameters. I haven't tested it so let me know if there's issues.  Also at the moment it's not possible to set the defaults for "Job" settings outside of code.

```python

def example_device_factory(default_config):
    """ Demonstrates how to create a preconfigured device.

    Returns
    -------
        result: Device
            A configured Device that the application can use.
    """
    from inkcut.device.plugin import Device, DeviceConfig
    from inkcut.device.protocols.hpgl import HPGLProtocol
    from inkcut.device.transports.serialport.plugin import SerialTransport, SerialConfig
    device =  Device(
        config=DeviceConfig(), # Pass any device config here
        connection=SerialTransport(
            config=SerialConfig(
                baudrate=115200, 
                port='/dev/ttyUSB0',
                # etc... Pass any serial config kwargs here
                ), 
            protocol=HPGLProtocol(), 
        ) 
    )
    
    return device

```

Then under the `DeviceDriver` of choice (or make a new entry) set `factory = example_device_factory`

frmdstryr, 2018-04-18:
Another possible option would be to configure inkcut on one system, copy the `json` config files to other systems. The config is saved under `/home/<username>/.config/inkcut`

srdco, 2018-04-21:
So working on device_factory setup.
Current device_factory is:
`def VisionVE810(default_config):
    """ Create a preconfigured device for Vision Engraver VE-810.
		(NOTE: Look at the following classes to determine available parameters:
		- Device Config -- inkcut/device/plugin.py --> DeviceConfig(Model)
		- Connection -- inkcut/device/transports/[type (e.g. serialport)]/plugin.py --> [TypeClass]Config(Model)
		

    Returns
    -------
        result: Device
            A configured Device that the application can use.
    """
    from inkcut.device.plugin import Device, DeviceConfig
    from inkcut.device.protocols.hpgl import HPGLProtocol
    from inkcut.device.transports.serialport.plugin import SerialTransport, SerialConfig
    device =  Device(
        config=DeviceConfig( # Pass any device config here
		    scale = [-1,1]
        ),
        connection=SerialTransport(
            config=SerialConfig(
                baudrate=119200, 
                port='/dev/ttyUSB0',
                rtscts=True
            )
        )
    )
    
    return device
`
But getting error:
`Traceback (most recent call last):
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/qt/qt_object_combo.py", line 117, in on_index_changed
    self.declaration.selected = item
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/declarative_meta.py", line 68, in declarative_change_handler
    engine.write(owner, change['name'], change)
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/expression_engine.py", line 217, in write
    pair.writer(owner, name, change)
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/standard_handlers.py", line 82, in __call__
    call_func(func, (), {}, scope)
  File "/home/sre/git/inkcut-20181417/inkcut/device/dialogs.enaml", line 60, in f
    dialog.device = plugin.get_device_from_driver(change['value'])
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/standard_tracer.py", line 61, in __call__
    engine.update(owner, self.name)
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/expression_engine.py", line 248, in update
    setattr(owner, name, pair.reader(owner, name))
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/standard_tracer.py", line 61, in __call__
    engine.update(owner, self.name)
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/expression_engine.py", line 248, in update
    setattr(owner, name, pair.reader(owner, name))
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/standard_handlers.py", line 101, in __call__
    return call_func(func, (tr,), {}, scope)
  File "/home/sre/git/inkcut-20181417/inkcut/device/dialogs.enaml", line 109, in <module>
    if t.id==connection.declaration.id][0] if connection else None
  File "/home/sre/git/inkcut-20181417/venv/lib/python3.6/site-packages/enaml/core/compiler_helpers.py", line 754, in wrapper
    return call_func(func, args, kwargs, scope)
  File "/home/sre/git/inkcut-20181417/inkcut/device/dialogs.enaml", line 109, in <listcomp>
    if t.id==connection.declaration.id][0] if connection else None
AttributeError: 'NoneType' object has no attribute 'id'
Fatal Python error: Aborted
`
This comes up when I switch to the Device in the Setup menu. Inkcut crashes with this error in the terminal. Any ideas?

srdco, 2018-04-21:
NOTE: 
The manifest entry is:
```
        DeviceDriver:
            manufacturer = 'Vision'
            model = ' VE-810 (XD) Engraver'
            width = '10in'
            length = '8in'
            protocols = ['hpgl']
            connections = ['serial']
            factory = VisionVE810
```
Not sure if the declarations are conflicting?

frmdstryr, 2018-04-21:
```python-traceback
  File "/home/sre/git/inkcut-20181417/inkcut/device/dialogs.enaml", line 109, in <listcomp>
    if t.id==connection.declaration.id][0] if connection else None
AttributeError: 'NoneType' object has no attribute 'id'
```
This looks like a bug... The problem is the UI is expecting the declarations to be used to create the connection object but in this case it's not happening because we're creating it manually. 

I do not like how I currently have this designed. I'm going to work on refactoring it so that this can all be done from the `default_config` parameter, like this.

```python

    DeviceDriver:
            manufacturer = 'Vision'
            model = ' VE-810 (XD) Engraver'
            width = '10in'
            length = '8in'
            protocols = ['hpgl']
            connections = ['serial']
            default_config = {
                'scale': [-1, 1],
                'connection': {
                    'serial': {
                        'baudrate': 119200,
                        'port': '/dev/ttyUSB0',
                        'rtscts': True,
                    }
                }
            }

```

It will also support 'job' and 'protocol' keys to pass defaults to those as well.

frmdstryr, 2018-09-03:
If this does not work please reopen. Thanks!

closed 2018-09-03 by frmdstryr