inkcut/inkcut #88: Crash when resize material if design too big

srdco, 2018-04-20:
Inkcut aborts without message when you try to resize material if the design is bigger than the new size.

Steps to recreate (with current inkcut/master):
1) Open a design
2) Click on Material, and set a width that is smaller than the graphic size
As soon as you tab, enter, or click out of the width box, Inkcut aborts.

Easy workaround is to resize the design smaller before you change the material size...

Also, is there no way to clear the work area back to a blank slate?

srdco, 2018-04-20:
Also, clicking "Read from Device" (with Vision Engraver selected, at least), causes an instant abort as well - even if the graphic size is small enough.

And ... I realized File --> Close clears the work area.

frmdstryr, 2018-04-20:
Thanks, I'll push an update to fix this. 
 
Regarding the "Read from Device" error, I'm going to remove this from the UI for now as it's not yet implemented. My device doesn't support it so I can't test it here, but I would like to add this for devices that do.  If someone knows what command is used to query the device retrieve this, that would be great!

frmdstryr, 2018-04-21:
This is now merged. If you get a chance try it out and let me know if you find any other issues.

srdco, 2018-04-22:
Just updated and tested it. Works great - Thanks!

closed 2018-04-22 by srdco