inkcut/inkcut #91: Project needs to be mentioned on Inkscape's website

Moini, 2018-04-25:
Hi there,

this project definitely needs to make it into the list at 
https://inkscape.org/en/gallery/=extension/ !

Can you make an account at inkscape.org and add a link to your website and extension?
Please also add the corresponding tags (follow link in description of linked page to learn how), to present it as well as possible.

Thanks!

raboof, 2018-04-25:
I guess we might first want to make a release (#78) and provide easy-to-follow installation instructions - but yes, agreed!

Moini, 2018-04-29:
Sounds good to me :-D

frmdstryr, 2018-09-03:
Thank you for creating this issue. 

I added the Inkcut for Inkscape extension to their extension list [here](https://inkscape.org/en/~frmdstryr/%E2%98%85inkcut). 

I assume it must be approved or something before it goes live onto the extension page?

closed 2018-09-03 by frmdstryr

Moini, 2018-09-03:
Thank you! I'm not sure for how long the gallery pages are cached, but that might explain if it took a couple of minutes to show in the list.