inkcut/inkcut #92: Debian packaging

raboof, 2018-04-25:
Aside from uploading to pypi (#78), it would be great if we could also publish .deb's.

raboof, 2018-04-26:
That would be great, and possibly also increase our chances of the package making it upstream!

raboof, 2018-05-26:
I made some tweaks to the deb-pkg branch, but now debuild for enaml fails for me with:

```
copying tools/vim/indent/enaml.vim -> enaml-0.10.2/tools/vim/indent
copying tools/vim/syntax/enaml.vim -> enaml-0.10.2/tools/vim/syntax
Writing enaml-0.10.2/setup.cfg
Creating tar archive
removing 'enaml-0.10.2' (and everything under it)
Traceback (most recent call last):
  File "setup.py", line 169, in <module>
    'develop': Develop},
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/stdeb/command/sdist_dsc.py", line 123, in run
    raise RuntimeError('original source dist cannot '
RuntimeError: original source dist cannot contain .pyc files
debuild: fatal error at line 660:
cannot find readable debian/changelog anywhere!
Are you in the source code tree?
Makefile:53: recipe for target 'enaml' failed
make: *** [enaml] Error 255
```

raboof, 2018-05-26:
Oh you already mentioned that in the PR at #93 , sorry

bianchifan, 2019-02-16:
ONe MORE for Debian packaging..
My Inet Connection is of limited bandwith, I had to go to an open accesspoint for installing.
The installion fetch more than 100 MB, although PYthon was installed in severeral versions.