inkcut/inkcut #105: Replace pyqtgraph with custom widget

frmdstryr, 2018-08-28:
pyqtgraph was used for simplicity but since all drawing is done with a QPainterPath, pyqtgraph and numpy can be removed.

frmdstryr, 2020-05-07:
For anyone that want's to try it, I just rebased this from a long time ago..
see https://github.com/inkcut/inkcut/tree/new-viewer

aaujon, 2020-05-25:
Hi, I give it a try, it works well, just 2 things : 
- when starting inkcut, the image is displayed too high, see screenshot  : 
[image]
- Sometimes, when moving the image (not sure how to reproduce it, just have to play with zoom and move) it crashes with : 
`2020-05-25 13:39:38,289 | DEBUG   | inkcut | Starting plugin 'inkcut.ui'
2020-05-25 13:39:38,290 | WARNING | inkcut | Plugin inkcut.ui state restored from: /home/iloarnaud/.config/inkcut/inkcut.ui.json
Traceback (most recent call last):
  File "/home/iloarnaud/.local/lib/python3.8/site-packages/enamlx/qt/qt_graphics_view.py", line 669, in mouseMoveEvent
  File "/home/iloarnaud/git/inkcut/inkcut/preview/view.enaml", line 59, in mouse_move_event
  File "/home/iloarnaud/.local/lib/python3.8/site-packages/enamlx/widgets/graphics_view.py", line 52, in __init__
NameError: name 'super' is not defined
`

aaujon, 2020-05-25:
Other stack when it crashes : 
2020-05-25 13:40:34,924 | WARNING | inkcut | Plugin inkcut.ui state restored from: /home/iloarnaud/.config/inkcut/inkcut.ui.json
Traceback (most recent call last):
  File "/home/iloarnaud/.local/lib/python3.8/site-packages/enamlx/qt/qt_graphics_view.py", line 663, in mousePressEvent
NameError: name 'super' is not defined
Traceback (most recent call last):
  File "/home/iloarnaud/.local/lib/python3.8/site-packages/enamlx/qt/qt_graphics_view.py", line 669, in mouseMoveEvent
  File "/home/iloarnaud/git/inkcut/inkcut/preview/view.enaml", line 59, in mouse_move_event
  File "/home/iloarnaud/.local/lib/python3.8/site-packages/enamlx/widgets/graphics_view.py", line 52, in __init__
TypeError: super() is not allowed in a declarative function,  use SomeClass.some_method(self, ...) instead.
Unhandled Python exception

frmdstryr, 2020-05-25:
Thanks for giving it a try! I think the errors above are related to #245 (seems like a problem with recent versions of enaml).

What are your thoughts on missing the x & y gridlines? Do you think those are those necessary?

aaujon, 2020-05-25:
The issue is different, but can be, I will try to downgrade it.
I  personally don't use the gridlines and I don't think they can be usefull in any workflow.

aaujon, 2020-05-25:
Ok the crash are caused by python 3.8/enaml 0.11/atom0.6, it's working great with python 3.7/enaml 0.10.4/atom0.4.2