inkcut/inkcut #110: Please add device

AdriVogel, 2018-09-26:
Please could you add another device to the list of supported devices for Inkcut? As I am quite ignorant about these technical issues, I hope I include the correct information.
ACS Eagle Computerised Cutter
SN: 0860006083
KnK Maxx
USB Port as well as serial port
Width - 28"
Length - Unlimited
Unfortunately don't know supported protocols

frmdstryr, 2018-09-26:
Thanks, I'll add it to the list so it's available in the text release.  Just to clarify can you specify the following:

manufacturer = 
model =

AdriVogel, 2018-10-29:
I see the issue was closed by accident. My apologies.  I am posting the question again:

Any idea when the next release will be done?
Manufacturer = KNK
Model = ACS Eagle/KNK Maxx - ACS Eagle was a special edition issued.
Width according to info on internet is 24" but when I measure the open width, it is 28"

frmdstryr, 2018-10-29:
Thanks, I added this device! Unsure of when the next release will be but you should be able to install the version from github by running `pip install git+https://github.com/codelv/inkcut.git`

AdriVogel, 2018-10-29:
Thank you! Much appreciated.

closed 2018-10-29 by AdriVogel