inkcut/inkcut #112: Add a new device, pls

cmic2, 2018-10-10:
manufacterer=roland
model=PNC-900
language=CAMM-GL III

(language as noted in the pdf user guide)

TYA
--
Michel Marcon (aka cmic)

frmdstryr, 2018-10-12:
Is this device working with Inkcut?

cmic2, 2018-10-13:
Hi

________________________________________
>De : frmdstryr <notifications@github.com>
>Envoyé : vendredi 12 octobre 2018 05:38
>À : codelv/inkcut
>Cc : cmic2; Author
>Objet : Re: [codelv/inkcut] Add a new device, pls (#112)
>
>Is this device working with Inkcut?

Not exactly. When I "Send to device", the Roland PNC900 doesn't move or clic. I suspect not to have a correct driver between Inkcut (or Linux ?) and the Roland cutter. 
I am trying to debug the serial interface /dev/ttyS0 with software tools (I don't have a serial Break box).

Any Hint ?  Or maybe I should try under Windows ? (Windows 10 for the matter, but no serial interface, only USB's)

Regards
--
michel marcon (aka cmic, retired sysadmin)

frmdstryr, 2018-10-13:
Linux will most likely be able to "talk to it" out of the box via serial / usb serial adapter. What needs done is a protocol needs written for the "CAMM-GL III" language so Inkcut sends the correct commands.  The [PNC900 manual](http://support.rolanddga.com/docs/documents/departments/technical%20services/manuals%20and%20guides/pnc900_user.pdf) lists all the commands in page 18. 

I'd prefer you or someone with the actual device give it a shot because it takes too much time pinging back and forth.

To do this:
1. Create a "CAMMGL3Protocol" in https://github.com/codelv/inkcut/blob/master/inkcut/device/protocols/camm.py
2. Override the methods to return the correct output . 
3. Once done, add it in the [manifest](https://github.com/codelv/inkcut/blob/master/inkcut/device/protocols/manifest.enaml)
4. Restart Inkcut then select the protocol in the Device -> Setup page 

You can also test sending commands using the Console, first connect using the button in the Control pane, then access the device object with these commands:

```python
from inkcut.core.workbench import InkcutWorkbench
workbench = InkcutWorkbench.instance()
device = workbench.get_plugin('inkcut.device').device
device.connection.write("The command")
```

cmic2, 2018-10-15:
Hi
________________________________________
>De : frmdstryr <notifications@github.com>
>Envoyé : samedi 13 octobre 2018 16:15
>À : codelv/inkcut
>Cc : cmic2; Author
>Objet : Re: [codelv/inkcut] Add a new device, pls (#112)
>
>Linux will most likely be able to "talk to it" out of the box via serial / usb serial adapter. What needs done is a protocol needs written for the >"CAMM-GL III" language so Inkcut sends the correct commands. The PNC900 manual<http://support.rolanddga.com/docs/documents>/departments/technical%20services/manuals%20and%20guides/pnc900_user.pdf> lists all the commands in page 18.
>
>I'd prefer you or someone with the actual device give it a shot because it takes too much time pinging back and forth.

OK. I get it. However, I only know C and Perl language (a bit of Forth ?). So I have to play Python? Interesting challenge.
Anyway. Thanks ever so much for this e-mail and hints. Think I'm gonna try...

Keep on keeping on your good work.
Cheers !
--
michel marcon (aka cmic http://wiki.cmic.be)

grany, 2019-01-05:
Hi,
the pnc-900 works well with the existing driver camm-gl1.

frmdstryr, 2019-01-05:
Thanks! I'll add it to the list

frmdstryr, 2019-01-05:
@cmic2 can you confirm if it works using camm-gl1 so this can be closed?

grany, 2019-01-05:
I have tested it with pnc-1000 and it works.

closed 2019-02-21 by frmdstryr