inkcut/inkcut #113: armhf Raspberry pi RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class

Joel-Mckay, 2018-10-29:
#install steps taken, as this step fails to build on Raspbian Stretch due to detected qt5 libs:
sudo pip install inkcut

#build steps
sudo apt-get install python3-pip  libcups2-dev
sudo apt-get install libxml2-dev libxslt-dev
sudo apt-get install python-qt4
sudo apt-get install libzbar-dev libzbarqt-dev libzbargtk-dev python-zbar
sudo apt-get install RPi.GPIO 
sudo apt-get install python-sip

git clone https://github.com/codelv/inkcut.git
cd inkcut
#build 32bit qt4 version
sudo pip uninstall inkcut
sudo pip install .

#also tried to force qt4 env of inkcut
qtchooser - qt4-arm-linux-gnueabihf  -run-tool=../../../../../usr/local/bin/inkcut
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Gtk-Message: Failed to load module "canberra-gtk-module"
Traceback (most recent call last):
  File "/usr/lib/arm-linux-gnueabihf/qt4/bin/../../../../../usr/local/bin/inkcut", line 11, in <module>
    load_entry_point('inkcut==2.0.6', 'console_scripts', 'inkcut')()
  File "/usr/local/lib/python2.7/dist-packages/inkcut/app.py", line 21, in main
    workbench.run()
  File "/usr/local/lib/python2.7/dist-packages/inkcut/core/workbench.py", line 120, in run
    ui = self.get_plugin('enaml.workbench.ui')
  File "/usr/local/lib/python2.7/dist-packages/enaml/workbench/workbench.py", line 152, in get_plugin
    plugin.start()
  File "/usr/local/lib/python2.7/dist-packages/enaml/workbench/ui/ui_plugin.py", line 56, in start
    self._create_application()
  File "/usr/local/lib/python2.7/dist-packages/enaml/workbench/ui/ui_plugin.py", line 208, in _create_application
    application = extension.factory()
  File "/usr/local/lib/python2.7/dist-packages/inkcut/core/manifest.enaml", line 27, in application_factory
    import qt4reactor
  File "/usr/local/lib/python2.7/dist-packages/qt4reactor.py", line 1, in <module>
    from qtreactor.pyqt4reactor import *
  File "/usr/local/lib/python2.7/dist-packages/qtreactor/pyqt4reactor.py", line 22, in <module>
    from qtreactor import qt4base
  File "/usr/local/lib/python2.7/dist-packages/qtreactor/qt4base.py", line 54, in <module>
    from PyQt4 import QtCore
RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class

#also tried to temporarily patch enaml.qt out to help narrow scope of issue(s)
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/joystick/plugin.py
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/job/models.py
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/core/svg.py
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/core/models.py
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/device/plugin.py
sed -i 's/from enaml.qt import/from PyQt4 import/g' inkcut/preview/plugin.py

#no luck so far, any hints would be welcome :-)

frmdstryr, 2018-10-29:
If you have PyQt5 installed, you can make it just use PyQt5. The check there now incorrectly assumes python 2 needs Qt4 when it works with either.

In `inkcut/core/manifest.enaml`  replace

```python
def application_factory():
    """ A factory function which creates the default Application
    and installs twisted support.
    """
    from enaml.qt.qt_application import QtApplication
    app = QtApplication()

    #: Install twisted support into the Qt event loop
    if sys.version_info.major == 2:
        try:
            import qt4reactor
        except ImportError:
            # Hack. What a stupid design
            from qtreactor import qtreactor_config
            qtreactor_config._instance.qtname = "PySide"
            from qtreactor import pyside4reactor as qt4reactor
        qt4reactor.install()
    else:
        import qt5reactor
        qt5reactor.install()

    return app

```

with just

```python
def application_factory():
    """ A factory function which creates the default Application
    and installs twisted support.
    """
    from enaml.qt.qt_application import QtApplication
    app = QtApplication()

    #: Install twisted support into the Qt event loop
    import qt5reactor
    qt5reactor.install()

    return app

```
Also make sure you `pip install qt5reactor`

frmdstryr, 2018-10-29:
I pushed a PR to fix this: #114

Can you try those changes and see if it works? It does the same thing.

Thanks!

Joel-Mckay, 2018-10-29:
Thanks for the reply, and the standalone inkcut now appears to load fine with qt5reactor on armhf/Rasbian(Debian Stretch).

#However, I also tested the plugins on the default inkscape 0.92.1
cd inkcut/plugins/inkscape
sudo cp inkcut* /usr/share/inkscape/extensions/
sudo chmod 755 /usr/share/inkscape/extensions/inkcut.py

"Cut selection..." does verify a path is selected, but does not seem to invoke a call.
"Open Current Document..." seems to delay awhile, and also seems to just exit.  

Cheers,
J

frmdstryr, 2018-10-29:
Sounds like inkcut isn't on the path or something. Does running the command `inkcut` launch it?  

You can look for logs under `~/.config/inkcut/logs` to see if it's exiting for some reason.

Joel-Mckay, 2018-10-29:
"Does running the command inkcut launch it? "
Yes (looks great by the way, I like the layout)

No plugin new log-events are committed to the text file in:
~/.config/inkcut/logs
or in:
~/.config/inkscape/extention-errors.log

Not really that helpful I'm afraid ;-)

frmdstryr, 2018-10-29:
You can try editing the `inkcut_cut.py` and changing the `            cmd = ['inkcut']` line to use the full path.

Also you can change `stdout=DEVNULL` to `stdout=subprocess.PIPE` and force `close_fds=False` so it stays open until inkcut is closed.

frmdstryr, 2018-10-30:
Ok, I guess since the extension never checks the output, the errors are just tossed out and it's exiting before logging is intialized.

Edit `inkcut/app.py` and replace it with this:

```python
# -*- coding: utf-8 -*-
"""
Copyright (c) 2017-2018, Jairus Martin.

Distributed under the terms of the GPL v3 License.

The full license is in the file LICENSE, distributed with this software.

Created on Jul 12, 2015

@author: jrm
"""
import os
import sys
import logging
import traceback

from logging.handlers import RotatingFileHandler

try:
    import faulthandler
    faulthandler.enable()
except:
    print("Warning: Faulthandler is missing!")

def init_logging():
    log_dir = os.path.expanduser('~/.config/inkcut/logs')
    if not os.path.exists(log_dir):
        os.makedirs(log_dir)
    log_filename = os.path.join(log_dir, 'inkcut.txt')
    log_format = '%(asctime)-15s | %(levelname)-7s | %(name)s | %(message)s'
    root = logging.getLogger()
    root.setLevel(logging.DEBUG)
    formatter = logging.Formatter(log_format)

    #: Log to stdout
    stream = logging.StreamHandler(sys.stdout)
    stream.setLevel(logging.DEBUG)
    stream.setFormatter(formatter)

    #: Log to rotating handler
    disk = RotatingFileHandler(
        log_filename,
        maxBytes=1024*1024*10,  # 10 MB
        backupCount=10,
    )
    disk.setLevel(logging.DEBUG)
    disk.setFormatter(formatter)

    root.addHandler(disk)
    root.addHandler(stream)
    print("Logging to {}".format(log_filename))
    return logging.getLogger('inkcut')

def main():
    log = init_logging()
    try:
        from inkcut.core.workbench import InkcutWorkbench
        workbench = InkcutWorkbench()
        workbench.run()
    except Exception as e:
        log.error(traceback.format_exc())
        raise

if __name__ == '__main__':
    main()

```

frmdstryr, 2018-10-30:
You should then (hopefully) see an error in `~/.config/inkcut/logs` that will help see what's going on.

Joel-Mckay, 2018-10-30:
Negative,
I attempted to run the patched incut with both the patched and un-patched inkcut_cut.py plugin.
same set off errs and empty logs were generated

#I have also verified the parameters array passed into inkcut from the plugin, and it refers to the path 
object name (I assume this is correct).
#... I suspect it has something to do with 0.92.1 inkscape... 
# So tried and failed with this patch to exclude a similar sounding possible silent bug.
=== modified file 'share/extensions/inkex.py'
--- share/extensions/inkex.py	2014-11-25 19:49:28 +0000
+++ share/extensions/inkex.py	2015-07-10 06:42:54 +0000
@@ -297,7 +297,7 @@
 
         svgwidth = self.document.getroot().get('width')
         viewboxstr = self.document.getroot().get('viewBox')
-        if viewboxstr:
+        if viewboxstr and svgwidth is not None:
             unitmatch = re.compile('(%s)$' % '|'.join(self.__uuconv.keys()))
             param = re.compile(r'(([-+]?[0-9]+(\.[0-9]*)?|[-+]?\.[0-9]+)([eE][-+]?[0-9]+)?)')

frmdstryr, 2018-10-30:
Ok, please close this and create a new issue. If possible attach the file that's not working.

Is it throwing any errors if you try to open it directly from Inkcut?