inkcut/inkcut #116: armhf Raspbian Stretch Inkscape 0.92.1 plugin issue with running inkcut

Joel-Mckay, 2018-10-30:
As per the other ticket: #113 
This is a rather evasive issue with the locally built install of inkcut... where the program works in stand-alone mode, but inkcut plugins do not complete the call from with Inkscape 0.92.1. 

Traceback (most recent call last):
File "inkcut_cut.py", line 73, in
effect.affect()
File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
self.effect()
File "inkcut_cut.py", line 67, in effect
close_fds=sys.platform != "win32")
File "/usr/lib/python2.7/subprocess.py", line 390, in init
errread, errwrite)
File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
raise child_exception
OSError: [Errno 9] Bad file descriptor

frmdstryr, 2018-10-30:
> I was able to see an error message with stdout=subprocess.STDOUT

Sorry, I did not realize yesterday that you changed it to `stdout=subprocess.STDOUT`  It should be `stdout=subprocess.PIPE`.  STDOUT is incorrect, hence the error. 

Please change the plugin to use that or revert back to the original plugin and pull the current master branch which should now collect logs correctly for startup failures.

Joel-Mckay, 2018-10-30:
I tried both setups attempting to narrow down where it is failing... Similar quiet failure mode in both cases.
 
I did attempt manually running the "inkcut open - --node 987sample" , and it waits for stdin data as expected.

Notably, in inkscape "open current document..." does appear to run inkcut in the process list for awhile as it tries to load. However, it also exits with no error messages of any kind.

In standalone mode, the inkcut Debug status messages filled the screen (needs a --verbose flag at some point), so I wrongly initially guessed this debug output may have been corrupting the data expected by file data parsers.  

nano inkcut/app.py
    #root.setLevel(logging.DEBUG)
    root.setLevel(logging.ERROR)
...
    #stream.setLevel(logging.DEBUG)
    stream.setLevel(logging.ERROR)
...
    #disk.setLevel(logging.DEBUG)
    disk.setLevel(logging.ERROR)
...
    #print("Logging to {}".format(log_filename))

This less verbose logging setup also does not show any Error messages in either the plugin, or running as standalone.

frmdstryr, 2018-10-30:
Can you post the logs from `~/.config/inkcut/logs`?

Joel-Mckay, 2018-10-30:
#inkcut in standalone mode with debug logging:

2018-10-30 14:52:52,731 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
2018-10-30 14:52:52,933 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
2018-10-30 14:52:58,465 | DEBUG   | inkcut | Inkcut loaded.
2018-10-30 14:52:58,579 | DEBUG   | inkcut | CLI | Commands loaded
2018-10-30 14:52:58,587 | DEBUG   | inkcut | CLI | No cli command was given.
2018-10-30 14:53:01,478 | DEBUG   | inkcut | CLI | Commands loaded
2018-10-30 14:53:01,485 | DEBUG   | inkcut | CLI | No cli command was given.
2018-10-30 14:53:03,586 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2018-10-30 14:53:03,606 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.count = 0'
2018-10-30 14:53:03,608 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.status = staged'
2018-10-30 14:53:03,609 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.units = in'
2018-10-30 14:53:03,610 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.started = None'
2018-10-30 14:53:03,611 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.speed = 0.0'
2018-10-30 14:53:03,611 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.ended = None'
2018-10-30 14:53:03,612 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.length = 0.0'
2018-10-30 14:53:03,613 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.duration = 0:00:00'
2018-10-30 14:53:03,614 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.progress = 0'
2018-10-30 14:53:03,615 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.data = '
2018-10-30 14:53:03,615 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.auto_approve = False'
2018-10-30 14:53:03,619 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.used = [0.0, 0.0]'
2018-10-30 14:53:03,620 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.force = 10'
2018-10-30 14:53:03,620 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.name = '
2018-10-30 14:53:03,621 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.color = #000000'
2018-10-30 14:53:03,621 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.use_speed = False'
2018-10-30 14:53:03,622 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.is_roll = False'
2018-10-30 14:53:03,623 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.padding = [10.0, 10.0, 10.0, 10.0]'
2018-10-30 14:53:03,662 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.cost = 1.0'
2018-10-30 14:53:03,663 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.use_force = False'
2018-10-30 14:53:03,664 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.speed = 10'
2018-10-30 14:53:03,665 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.size = [1800.0, 2700.0]'
2018-10-30 14:53:03,669 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.info = <inkcut.job.models.JobInfo object at 0x6a6fe7b0>'
2018-10-30 14:53:03,670 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.scale = [1.0, 1.0]'
2018-10-30 14:53:03,672 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.copy_spacing = [10.0, 10.0]'
2018-10-30 14:53:03,674 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.auto_rotate = False'
2018-10-30 14:53:03,676 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.auto_copies = False'
2018-10-30 14:53:03,677 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.plot_weedline_padding = [10.0, 10.0, 10.0, 10.0]'
2018-10-30 14:53:03,679 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.plot_weedline = False'
2018-10-30 14:53:03,681 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.material = <inkcut.job.models.Material object at 0x6a7a5670>'
2018-10-30 14:53:03,683 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.copy_weedline_padding = [10.0, 10.0, 10.0, 10.0]'
2018-10-30 14:53:03,685 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.copies = 1'
2018-10-30 14:53:03,687 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.auto_scale = False'
2018-10-30 14:53:03,689 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.align_center = [False, False]'
2018-10-30 14:53:03,690 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.mirror = [False, False]'
2018-10-30 14:53:03,691 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.feed_to_end = False'
[... cut ...]
Requirement already satisfied: ptyprocess>=0.5 in /usr/local/lib/python2.7/dist-packages (from pexpect; sys_platform != "win32"->ipython>=4.0.0->ipykernel>=4.1->qtconsole->inkcut==2.0.7.dev0)
Requirement already satisfied: scandir; python_version < "3.5" in /usr/local/lib/python2.7/dist-packages (from pathlib2; python_version == "2.7" or python_version == "3.3"->ipython>=4.0.0->ipykernel>=4.1->qtconsole->inkcut==2.0.7.dev0)
Installing collected packages: inkcut
  Running setup.py install for inkcut ... done
Successfully installed inkcut-2.0.7.dev0

frmdstryr, 2018-10-31:
Thanks. Unfortunately I don't see any errors. 

Can the file which you're trying to use be opened directly in Inkcut?

Also if you remove the arguments to open the file altogether and run the extension does Inkcut open?

Joel-Mckay, 2018-10-31:
"Can the file which you're trying to use be opened directly in Inkcut?"
Yes, the log should show the simple drawing.svg file being opened and closed.  
I also tried inkex.debug(inkex.etree.tostring(document)) which does show p.stdin.write(...) normally does get something that looks like a properly formatted svg string.

"Also if you remove the arguments to open the file altogether and run the extension does Inkcut open?"
Same silent log output on both plugin calls. 
I also tried the inkcut_cut.py plugin debug mode to force specific python versions in the call.  

I have included the shell log below too which includes some additional Gtk messages.
---
Logging to /home/pi/.config/inkcut/logs/inkcut.txt
2018-10-30 14:52:52,731 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
2018-10-30 14:52:52,933 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Gtk-Message: Failed to load module "canberra-gtk-module"
2018-10-30 14:52:58,465 | DEBUG   | inkcut | Inkcut loaded.
2018-10-30 14:52:58,579 | DEBUG   | inkcut | CLI | Commands loaded
2018-10-30 14:52:58,587 | DEBUG   | inkcut | CLI | No cli command was given.
2018-10-30 14:53:01,478 | DEBUG   | inkcut | CLI | Commands loaded
2018-10-30 14:53:01,485 | DEBUG   | inkcut | CLI | No cli command was given.
2018-10-30 14:53:03,586 | DEBUG   | inkcut | Starting plugin 'inkcut.job'
2018-10-30 14:53:03,606 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.count = 0'
2018-10-30 14:53:03,608 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.status = staged'
2018-10-30 14:53:03,609 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.units = in'
2018-10-30 14:53:03,610 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.started = None'
2018-10-30 14:53:03,611 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.speed = 0.0'
2018-10-30 14:53:03,611 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.ended = None'
2018-10-30 14:53:03,612 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.length = 0.0'
2018-10-30 14:53:03,613 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.duration = 0:00:00'
2018-10-30 14:53:03,614 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.progress = 0'
2018-10-30 14:53:03,615 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.data = '
2018-10-30 14:53:03,615 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.JobInfo object at 0x6a6fe7b0>.auto_approve = False'
2018-10-30 14:53:03,619 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.used = [0.0, 0.0]'
2018-10-30 14:53:03,620 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.force = 10'
2018-10-30 14:53:03,620 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.name = '
2018-10-30 14:53:03,621 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.color = #000000'
2018-10-30 14:53:03,621 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.use_speed = False'
2018-10-30 14:53:03,622 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.is_roll = False'
2018-10-30 14:53:03,623 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.padding = [10.0, 10.0, 10.0, 10.0]'
2018-10-30 14:53:03,662 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.cost = 1.0'
2018-10-30 14:53:03,663 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.use_force = False'
2018-10-30 14:53:03,664 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.speed = 10'
2018-10-30 14:53:03,665 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Material object at 0x6a7a5670>.size = [1800.0, 2700.0]'
2018-10-30 14:53:03,669 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.info = <inkcut.job.models.JobInfo object at 0x6a6fe7b0>'
2018-10-30 14:53:03,670 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.scale = [1.0, 1.0]'
2018-10-30 14:53:03,672 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.copy_spacing = [10.0, 10.0]'
2018-10-30 14:53:03,674 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.auto_rotate = False'
2018-10-30 14:53:03,676 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.auto_copies = False'
2018-10-30 14:53:03,677 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.plot_weedline_padding = [10.0, 10.0, 10.0, 10.0]'
2018-10-30 14:53:03,679 | DEBUG   | inkcut | Restoring state '<inkcut.job.models.Job object at 0x6a6fe990>.plot_weedline = False'
[... cut ...]
2018-10-30 14:53:32,663 | INFO    | inkcut | {'value': <inkcut.job.models.Job object at 0x6281f4b8>, 'object': <inkcut.job.plugin.JobPlugin object at 0x6a7a1eb8>, 'type': 'update', 'name': 'job', 'oldvalue': <inkcut.job.models.Job object at 0x6a6fe990>}
2018-10-30 14:53:32,791 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2018-10-30 14:53:32,840 | INFO    | inkcut | Saving state due to change: {'value': <inkcut.job.models.Job object at 0x6281f4b8>, 'object': <inkcut.job.plugin.JobPlugin object at 0x6a7a1eb8>, 'type': 'update', 'name': 'job', 'oldvalue': <inkcut.job.models.Job object at 0x6a6fe990>}

frmdstryr, 2018-10-31:
The log does not seem to indicate it's opening from stdin. If I run

```bash
 cat tests/data/arc.svg | inkcut open -
```

I see output saying

```
2018-10-31 14:24:23,205 | DEBUG   | inkcut | CLI | Runinng command 'open' with args: Namespace(cmd=<inkcut.cli.plugin.Command object at 0x7f4287853d48>, nodes=None, path='-')
2018-10-31 14:24:23,206 | DEBUG   | inkcut | Opening document from stdin...
2018-10-31 14:24:23,206 | INFO    | inkcut | Closing -
```

and see the file. 

I do use Inkcut regularly with a raspberry pi but use xfce and haven't been using the inkscape plugin as I typically just open the file directly.

Joel-Mckay, 2018-11-04:
I thought it may be inkscape 0.92.1 api, and built 0.92.3 to test the plug-in again... but it seems they changed quite a bit in the next release:
---
Traceback (most recent call last):
  File "inkcut_open.py", line 27, in <module>
    inkex.localize()
TypeError: 'module' object is not callable

/usr/local/share/inkscape/extensions/inkex/deprecated.py:67: DeprecationWarning: Effect.affect is now `Effect.run()` with the same args
  warnings.warn(msg, DeprecationWarning)
  
/usr/local/share/inkscape/extensions/inkex/deprecated.py:67: DeprecationWarning: Effect.selected is now a dictionary in the svg document. Use self.svg.selected instead.
  warnings.warn(msg, DeprecationWarning)
---
Unfortunately, I think the standalone-inkcut mode will be the only mode for this system for now ;-)

Thanks for the advice, and the inkcut program itself still seems to load the file fine. 
---
$ cat drawing.svg | inkcut open -
Logging to /home/pi/.config/inkcut/logs/inkcut.txt
2018-11-04 14:24:59,740 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
2018-11-04 14:24:59,924 | INFO    | root | Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Gtk-Message: Failed to load module "canberra-gtk-module"
2018-11-04 14:25:04,743 | DEBUG   | inkcut | Inkcut loaded.
2018-11-04 14:25:04,858 | DEBUG   | inkcut | CLI | Commands loaded
2018-11-04 14:25:07,943 | DEBUG   | inkcut | CLI | Commands loaded
2018-11-04 14:25:11,994 | DEBUG   | inkcut | Starting plugin 'inkcut.device'
2018-11-04 14:25:12,004 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.commands_disconnect = '
2018-11-04 14:25:12,005 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.custom_rate = -1.0'
2018-11-04 14:25:12,005 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.force_units = g'
2018-11-04 14:25:12,006 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.scale = [1.0, 1.0]'
2018-11-04 14:25:12,007 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.force = 40.0'
2018-11-04 14:25:12,008 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.speed_units = in/s'
2018-11-04 14:25:12,008 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.force_enabled = False'
2018-11-04 14:25:12,009 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.commands_before = '
2018-11-04 14:25:12,009 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.speed_enabled = False'
2018-11-04 14:25:12,010 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.interpolate = False'
2018-11-04 14:25:12,011 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.spooled = False'
2018-11-04 14:25:12,011 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.step_size = 3.5433070866'
2018-11-04 14:25:12,012 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.sample_rate = 100'
2018-11-04 14:25:12,012 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.test_mode = False'
2018-11-04 14:25:12,013 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.commands_after = '
2018-11-04 14:25:12,013 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.step_time = 35.0'
2018-11-04 14:25:12,014 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.rotation = 0'
2018-11-04 14:25:12,015 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.speed = 4.0'
2018-11-04 14:25:12,015 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.commands_connect = '
2018-11-04 14:25:12,016 | DEBUG   | inkcut | Restoring state '<inkcut.device.plugin.DeviceConfig object at 0x6485c1c0>.absolute = False'
2018-11-04 14:25:12,110 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.parity = None'
2018-11-04 14:25:12,111 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.baudrate = 9600'
2018-11-04 14:25:12,111 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.bytesize = 8'
2018-11-04 14:25:12,112 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.xonxoff = False'
2018-11-04 14:25:12,113 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.rtscts = False'
2018-11-04 14:25:12,113 | DEBUG   | inkcut | Restoring state '<inkcut.device.transports.serialport.plugin.SerialConfig object at 0x6a76bc10>.stopbits = 1'
[... cut ...]
2018-11-04 14:25:21,816 | INFO    | inkcut | Saving state due to change: {'type': 'request'}
2018-11-04 14:25:21,870 | INFO    | inkcut | Saving state due to change: {'value': [<inkcut.job.models.Job object at 0x64860800>, <inkcut.job.models.Job object at 0x64860990>, <inkcut.job.models.Job object at 0x648607b0>], 'object': <inkcut.job.plugin.JobPlugin object at 0x6485c260>, 'type': 'update', 'name': 'jobs', 'oldvalue': [<inkcut.job.models.Job object at 0x64860800>, <inkcut.job.models.Job object at 0x64860990>]}