inkcut/inkcut #117: Problems brewing with CUPS how about direct access to /dev/lp0

jimthouston, 2018-11-01:
I have been playing with a New Hermes Vanguard engraver.  It works if I save the generated HPGL to a file and cat the file to /dev/lp0.

I spent a few hours trying to get CUPS on Ubuntu 18.04 to let me configure a raw generic printer.  The Ubuntu settings printer configuration didn't support a printer on /dev/lp0.
Using the CUPS web server at http://localhost:631 let me use /dev/lp0 but it didn't offer a raw or generic option.  I was able to find a .ppd file which almost worked.

It looks like the CUPS developers have deprecated the generic and raw interfaces.  See this page:
https://www.cups.org/blog/2018-06-06-demystifying-cups-development.html

frmdstryr, 2018-11-01:
If cat works, you could try just hardcoding https://github.com/codelv/inkcut/blob/master/inkcut/device/transports/disk/plugin.py#L55 to use `path = self.path = '/dev/lp0'` then select the `Save to file` connection type and see if it works.

jimthouston, 2018-11-03:
I found the lpr option "-o raw" which worked to send the file to the engraver without modifications.  So I'm a step closer to making the print spooler work.

frmdstryr, 2018-11-03:
Very good. Is your engraver 2.5D or 3D?

I've been retrofitting an old [Dahlgren System 300z](https://www.codelv.com/blog/2018/8/retrofitting-an-engraver-part-1) and am planning to add some engraving features to Inkcut (like spindle control, setting Z value, etc..). So I'm open to suggestions on what to add.

jimthouston, 2018-11-03:
I'm using the controller which came with my engraver so it is only 2D.  I bought it thinking of it as a small milling machine but I decided to see what the existing controller would do.  I have the Win95 opensys software that originally ran it.  I captured some data so I know some of the commands which are used to set the feed rate and depth.  I have not tried adding them to inkcut yet.  I have been doing the setup from the front panel buttons on the controller.

I did a quick change to the printer/plugin.py to add a "-o raw" option to lpr and was able to send a useful file to the engraver.

frmdstryr, 2018-11-03:
Great!  If that's an options that others might need it could be added to the printer config object and a checkbox added to set it from the UI.

raboof, 2018-11-05:
CUPS is pretty much overkill anyway, so I agree some UI to write directly to the printer device file would be helpful.

Did "write to file" with `/dev/lp0` as the target file work for you @jimthouston ?

jimthouston, 2018-11-05:
Yes changing the disk/plugin.py to hard code /dev/usb/lp1 worked.  I also changed the permissions so I could write to the printer.  I was also able to use the write to file interface without modification by putting
setting the format to lp1 and the directory to /dev/usb.

It took me a bit to understand the format/directory split.  I would have expected that I could browse to a filename rather than just a directory.