inkcut/inkcut #118: Refine RH360 - bumping into casing

bueffeljaeger, 2018-11-03:
Hi,
Yesterday I got my refine RH360 and it's my first time using inkcut. But whenever I want to run a cut the cutting carriage moves all the way to the left side until it bumps into the casing. Then I have to cut-off the power supply.

I know there are a view users out there running inkcut in combination with the RH360, so maybe someone can tell me the correct setting. In my device settings, I do have used the driver for "US Cutter Refine MH-721" (as mentioned somewhere else) width: 10in and length: 20in. The protocol language is set to DMPL, mode 1.

Here is a screenshot of the GUI:
[image]

Any idea what might be the problem?

frmdstryr, 2018-11-03:
Can you try using HPGL?

bueffeljaeger, 2018-11-03:
I did try HPGL but then the cutter is not moving at all.

According to the documentation it only supports DMPL. I could get the cutter working properly using ArtCut 2009 in VMWare, where I selected the same printer (MH-721). But I really want to use it in linux because this is my main OS.

frmdstryr, 2018-11-03:
Ok, there is probably issues with the [dmpl](https://github.com/codelv/inkcut/blob/master/inkcut/device/protocols/dmpl.py) protocol. 

I never used it but simply ported it from work lwernars added to the v1.0 inkcut. 

You can test sending commands directly by opening the Console and using

```python
from inkcut.core.workbench import InkcutWorkbench
# Get device handle
device = InkcutWorkbench.instance().get_plugin('inkcut.device').device
device.connect() # Or click the connect button in the Control tab
device.connection.write("D100,100")
```

bueffeljaeger, 2018-11-04:
OK, if just put in the commands:

```
In [1]: from inkcut.core.workbench import InkcutWorkbench
In [2]: device = InkcutWorkbench.instance().get_plugin('inkcut.device').device
In [3]: device.connect()
Out[3]: <Deferred at 0x7fb37d0791d0 current result: None>
In [4]: device.connection.write("D100,100")
```
With the write command, the cutter moved down a bit.

bueffeljaeger, 2018-11-04:
OK, I made an other test.

I did draw a rectangle 20x20mm (from origin) in artcut and saved to output to a file.
The generated content is this:

`;:H A L0 ECN U V10 U0,0 D0,0 D0,800,800,800,800,0,0,0 U0,0 U800,0 U F800 @ `

Then I draw the same rectangle in inkscape, saved it as SVG and opened it it inkcut:

[image]

As shown in the preview, the rectangle is 0.78x0.78in, which is 20x20mm. Then I saved the output (dmpl) to a file, which ended up in this result:

` ;:H A L0  U0,70.82814073160743  D70.82813973214373,70.82814073160743  D70.82813973214373,0  D0,0  D0,70.82814073160743  U0,0 `

Btw, I changed the DMPL protocol mode in inkcut to 3, which seemed to match most with the artcut output.

bueffeljaeger, 2018-11-04:
One more thing. If I connect to my plotter via minicom and send command by command everything works fine until I send the first "D" command with the comma ... then one axis of the printer starts running and never stops again.

Then I tried to reduce the amount of numbers after the comma. And I guess that's the problem. My cutter starts going crazy if there are more then 4 decimal places.

If you can point out where the calculation is done within inkcut, I'll try to change it (must be some kind of fprintf) and see how it works.

I am pretty sure the cutout will be not in scale (I guess factor 11.29) but at least it shout cut out something

frmdstryr, 2018-11-04:
Thanks for investigating.

 HPGL scales the move arguments and casts them to an int see https://github.com/codelv/inkcut/blob/master/inkcut/device/protocols/hpgl.py so DMPL may require the same thing.

bueffeljaeger, 2018-11-05:
Bingo. That's it ... now it works like a charm. Here are my changes to dmpl.py

```
10c10
< from atom.api import Enum, Instance
---
> from atom.api import Enum, Instance, Float, Int
20d19
< 
23c22,23
< 
---
>     scale = Float(1021/90.0)
>     
35a36
>         x, y = int(x*self.scale), int(y*self.scale)
```

frmdstryr, 2018-11-06:
Great! Would you be able to do a pull request?

frmdstryr, 2019-02-01:
Thanks for investigating this. I've merged in your patch.

closed 2019-02-01 by frmdstryr