inkcut/inkcut #119: Error linuxmint 19

chetosvsgeez, 2018-11-05:
compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-nqjbc4rl/twisted/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-8zg3bzm4-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-nqjbc4rl/twisted/

raboof, 2018-11-05:
can you share the command you used and the full output?

chetosvsgeez, 2018-11-05:
pip3 install inkcut
Collecting inkcut
  Using cached https://files.pythonhosted.org/packages/71/4b/ad44a71e9fc55fec8d44f9e7f9f2773fe8a249d4bdb7ed7dda7dc80353bd/inkcut-2.0.6-py2.py3-none-any.whl
Collecting pyqtgraph (from inkcut)
  Using cached https://files.pythonhosted.org/packages/cd/ad/307e0280df5c19986c4206d138ec3a8954afc722cea991f4adb4a16337d9/pyqtgraph-0.10.0.tar.gz
Collecting pyserial>=3.4 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting enaml>=0.10 (from inkcut)
  Using cached https://files.pythonhosted.org/packages/31/47/d3b872cab4b6e02c9cc17f820ec5f020e59ac3a77b96b147f41ab11f30ac/enaml-0.10.2-cp36-cp36m-manylinux1_x86_64.whl
Collecting enamlx (from inkcut)
  Using cached https://files.pythonhosted.org/packages/be/5a/217fda69ee0e5fa2a9a558de29ead1591e6861b0b6869a075c1346810275/enamlx-0.3.0-py2.py3-none-any.whl
Collecting qt5reactor; python_version >= "3.0" (from inkcut)
  Using cached https://files.pythonhosted.org/packages/f4/0e/a4248c0e3c712db8ffb1efde76b0b609012af0b74803e901a7e2847d88bc/qt5reactor-0.5-py2.py3-none-any.whl
Collecting twisted (from inkcut)
  Using cached https://files.pythonhosted.org/packages/5d/0e/a72d85a55761c2c3ff1cb968143a2fd5f360220779ed90e0fadf4106d4f2/Twisted-18.9.0.tar.bz2
Collecting pycups; sys_platform == "linux" (from inkcut)
  Using cached https://files.pythonhosted.org/packages/00/ce/836a0deb8b24bcd5f850f8fb97f99fb4abd7374e078b9e6df5a0838f8eb5/pycups-1.9.74.tar.bz2
Collecting qtconsole (from inkcut)
  Using cached https://files.pythonhosted.org/packages/37/22/0d8474f78a8c421d485ac2339de7c871d535160f09f170de90c8185b87c4/qtconsole-4.4.2-py2.py3-none-any.whl
Collecting jsonpickle (from inkcut)
  Using cached https://files.pythonhosted.org/packages/ca/ce/97404d5aeb58e6155c216825c81b50f6eca8a5345c582317ae48391878f8/jsonpickle-1.0-py2.py3-none-any.whl
Collecting lxml (from inkcut)
  Using cached https://files.pythonhosted.org/packages/03/a4/9eea8035fc7c7670e5eab97f34ff2ef0ddd78a491bf96df5accedb0e63f5/lxml-4.2.5-cp36-cp36m-manylinux1_x86_64.whl
Collecting PyQt5; python_version >= "3.0" (from inkcut)
  Using cached https://files.pythonhosted.org/packages/d4/bf/d884da8e2f7096d201c891d515eb6813a8e85df5eb6f5e12e867bf1d831c/PyQt5-5.11.3-5.11.2-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl
Collecting numpy (from pyqtgraph->inkcut)
  Using cached https://files.pythonhosted.org/packages/ff/7f/9d804d2348471c67a7d8b5f84f9bc59fd1cefa148986f2b74552f8573555/numpy-1.15.4-cp36-cp36m-manylinux1_x86_64.whl
Collecting kiwisolver>=1.0.0 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/69/a7/88719d132b18300b4369fbffa741841cfd36d1e637e1990f27929945b538/kiwisolver-1.0.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting setuptools (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/82/a1/ba6fb41367b375f5cb653d1317d8ca263c636cff6566e2da1b0da716069d/setuptools-40.5.0-py2.py3-none-any.whl
Collecting atom>=0.4.1 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/8f/05/e30d7211a51ad4f59b85b6847eb06f1ad6bb0fa7c6e70d65d9dcb50fd8ab/atom-0.4.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting qtpy>=1.3 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/8e/dd/eef9ac8a6b30cb38fedd1187848d9f3e39b02e5c29e684130c85f314cea6/QtPy-1.5.2-py2.py3-none-any.whl
Collecting ply>=3.4 (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/a3/58/35da89ee790598a0700ea49b2a66594140f44dec458c07e8e3d4979137fc/ply-3.11-py2.py3-none-any.whl
Collecting future (from enaml>=0.10->inkcut)
  Using cached https://files.pythonhosted.org/packages/90/52/e20466b85000a181e1e144fd8305caf2cf475e2f9674e797b222f8105f5f/future-0.17.1.tar.gz
Collecting zope.interface>=4.4.2 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/19/17/1d198a6aaa9aa4590862fe3d3a2ed7dd808050cab4eebe8a2f2f813c1376/zope.interface-4.6.0-cp36-cp36m-manylinux1_x86_64.whl
Collecting constantly>=15.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/b9/65/48c1909d0c0aeae6c10213340ce682db01b48ea900a7d9fce7a7910ff318/constantly-15.1.0-py2.py3-none-any.whl
Collecting incremental>=16.10.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/f5/1d/c98a587dc06e107115cf4a58b49de20b19222c83d75335a192052af4c4b7/incremental-17.5.0-py2.py3-none-any.whl
Collecting Automat>=0.3.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/a3/86/14c16bb98a5a3542ed8fed5d74fb064a902de3bdd98d6584b34553353c45/Automat-0.7.0-py2.py3-none-any.whl
Collecting hyperlink>=17.1.1 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/a7/b6/84d0c863ff81e8e7de87cff3bd8fd8f1054c227ce09af1b679a8b17a9274/hyperlink-18.0.0-py2.py3-none-any.whl
Collecting PyHamcrest>=1.9.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/9a/d5/d37fd731b7d0e91afcc84577edeccf4638b4f9b82f5ffe2f8b62e2ddc609/PyHamcrest-1.9.0-py2.py3-none-any.whl
Collecting attrs>=17.4.0 (from twisted->inkcut)
  Using cached https://files.pythonhosted.org/packages/3a/e1/5f9023cc983f1a628a8c2fd051ad19e76ff7b142a0faf329336f9a62a514/attrs-18.2.0-py2.py3-none-any.whl
Collecting pygments (from qtconsole->inkcut)
[... cut ...]
    Exception: It appears another version of pyqtgraph is already installed at /home/gerardo/.local/lib/python3.6/site-packages/pyqtgraph; remove this before installing.
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-u5y2ed2n/pyqtgraph/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-lb5fz8yh-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-u5y2ed2n/pyqtgraph/

frmdstryr, 2018-11-08:
> error: invalid command 'bdist_wheel'

This means [wheel](https://stackoverflow.com/questions/34819221/why-is-python-setup-py-saying-invalid-command-bdist-wheel-on-travis-ci/44862371#44862371) is not installed. Please run `pip3 install wheel` and try again

closed 2018-12-21 by frmdstryr

sudo-joseph, 2019-11-10:
Ran into same issue and installing wheels solved. Thank you!