inkcut/inkcut #120: Add supported device, Secabo C60ii

Marzel86, 2018-11-10:
Hello, 
First of all thank you for developing inkcut! It's the only way to get my cheap Chinese Printer running.

Brand: Secabo
Model: C60ii
Protocoll: HPGL
Connection type: USB to Serial, Baudrats 9600, Bytesize 8, Parity none, Stop Bit 1, RTS/CTS checked, XON/XOFF checked
Widht: 60cm
Length: i Think endless

It works, but there are 2 restrictions:
- the printer hasn't a firmware blade offset correction - small prints are looking bad
- after the successfull print the plotter is displaying "waiting..." and do not respond to any input. If you start the print in inkcut again nothing happens but the plotter switched back "on line" and is responding.

Can you give me a hint how to handle this issues?

frmdstryr, 2018-11-12:
Thanks for using Inkcut!  I added this device to the list so it will be in the next release.

> the printer hasn't a firmware blade offset correction - small prints are looking bad

Yes, this is known see #76 . Perhaps someone can add contribute this as it's not something I need.

> after the successful print the plotter is displaying "waiting..." and do not respond to any input. If you start the print in inkcut again nothing happens but the plotter switched back "on line" and is responding

It sounds like this cutter expects some command to tell it that it's finished.  

You can try:

1. Go to "Device -> Setup"
2. Open the "Device" tab
3. Scroll down to the "Job commands" and  `IN;` to the After job

The "IN;" command should "initialize the device".

Marzel86, 2018-12-15:
Today I used my cutter again and tried to add the "IN;" command --> now it works!
Thank you very much. :)

I also tried the hack described in #76 and used the inkscape hpgl export to ge the blade offset correction. Works for me.

As mentioned in #61 it would be great having the opportunity to change to "mm" instead of "in".

raboof, 2018-12-15:
Could we add this as a preset?

frmdstryr, 2018-12-15:
I think it'd be fine.  Just add a  [`finish`](https://github.com/codelv/inkcut/blob/master/inkcut/device/plugin.py#L195) method in in [hpgl.py](https://github.com/codelv/inkcut/blob/master/inkcut/device/protocols/hpgl.py)

kritzikratzi, 2024-07-27:
i know this is closed, but on the secabo c30iii i find that this works much better than "IN;": 

    U U0,0 @;

this lifts the pen, moves it to 0,0, and stops the print job.