inkcut/inkcut #121: Installation on Ubuntu 18.04 call to pyside failed

1-prop-head, 2018-12-06:
on initial start of inkcut from command line the main inkcut window did not appear but did have the following in a window.

An error prevented startup. This is either a bug or a required library is missing on your system. 

File "/usr/local/lib/python2.7/dist-packages/inkcut/job/plugin.py", line 151, in _refresh_preview
    pen=plot.pen_device,
  File "/usr/local/lib/python2.7/dist-packages/inkcut/preview/plugin.py", line 47, in _default_pen_device
    return pg.mkPen((235, 194, 194), style=QtCore.Qt.DashLine)
  File "/home/dm/.local/lib/python2.7/site-packages/pyqtgraph/functions.py", line 293, in mkPen
    pen.setStyle(style)
TypeError: 'PySide.QtGui.QPen.setStyle' called with wrong argument types:
  PySide.QtGui.QPen.setStyle(PenStyle)
Supported signatures:
  PySide.QtGui.QPen.setStyle(PySide.QtCore.Qt.PenStyle)

Please report this issue or request help at https://github.com/codelv/inkcut

frmdstryr, 2018-12-06:
Can you try with PyQt / PyQt5?

1-prop-head, 2018-12-07:
installed via app install python.pyqt5
reboot
from terminal run inkcut at command prompt
retry and get identical Workspace startup error dialog as above

frmdstryr, 2018-12-07:
You might need to uninstall PySide if it's still using that (and the pyside system package if that's installed).

1-prop-head, 2018-12-07:
If I uninstall pyside when I invoke inkcut at the command prompt I am immediately told that pyside is missing.
If I install pyside I am then give the dialog provided above.  Am I missing some extra component of pyside or some other feature.

Thanks for your help - I would be happy to load a different linux that you recommend since the laptop will be dedicated to operation of a Roland GX-24 printer 

This is what is provided in the terminal when inkcut is invoked and provides the above dialog

Current thread 0x00007ff8f1c77740 <inkcut> (most recent call first):
  File "/home/dm/.local/lib/python2.7/site-packages/enaml/qt/qt_application.py", line 48 in start
  File "/home/dm/.local/lib/python2.7/site-packages/enaml/workbench/ui/ui_plugin.py", line 105 in start_application
  File "/usr/local/lib/python2.7/dist-packages/inkcut/core/workbench.py", line 128 in run
  File "/usr/local/lib/python2.7/dist-packages/inkcut/app.py", line 21 in main
  File "/usr/local/bin/inkcut", line 11 in <module>
Aborted (core dumped)

frmdstryr, 2018-12-07:
Can you trying installing PyQt5 with pip? `pip install PyQt5`

frmdstryr, 2018-12-21:
Using PySide is not recommended. Please use PyQt or PyQt5 and reopen if needed.

closed 2018-12-21 by frmdstryr