inkcut/inkcut #130: Support for legacy Cricut devices

untitled01exe, 2019-01-17:
Cricut has discontinued older machine software completely. 
The Cricut Professional, Create, Expression, Expression 2, and Mini no longer have any supported software, or third party software support. 
Lets create a home for these machines on Inkcut!

frmdstryr, 2019-01-17:
A quick search brought up https://github.com/Arlet/Freecut and a few links on there about Circut I found http://oscopetutorial.com/cricut/index.php?title=Main_Page which has python code for talking with the Circut.  If that works it should be a matter of adding a Inkcut driver for it.

TrendMend, 2019-05-30:
Hey there! I own a cricut expression, and I installed a new firmware on it, called FreeExpression, that is a fork of FreeCut. It supports hpgl, so (I am assuming) inkcut will work with it great, otherwise, if you want to not overwrite the original firmware, you can use libcutter (but it isn't that good). I haven't checked out inkcut yet, but if it can take plugins, I am sure someone can make one. The only issue is that ProvoCraft is really sucky when it comes to this, and there have been many many lawsuits over this, so I don't think it could be something the developers implement, but something you would have to work for.

jschwalbe, 2019-10-06:
@TrendMend you mentioned elsewhere that you got it working with Inkcut and that you actually prefer it? Any instructions to make it work?

TrendMend, 2019-10-06:
Yeah, I do use the send to feature, but it just took some trial an error in
the settings, unfortunately it is lost to time thanks to my dead laptop
drive, but if you check some of my comments you should find the scale
amount (to infinity) that is the most important aspect of all

jschwalbe, 2019-10-07:
Haven't been able to figure out how to add it as a device yet. Tricky tricky :)

charely6, 2022-08-05:
did you ever get this to work?