inkcut/inkcut #131: Brother Designncut DC100

DrDabadu, 2019-01-20:
Is it possible to add support for the Brother Design n cut (DC 100), might be DC 200 outside Europe.

Thanks for any suggestion

frmdstryr, 2019-01-21:
Thanks for your interest. I don't know much about this device. 

To add support someone needs to provide technical details on:

- the communication interface (ex usb), type (serial, ethernet, etc..), and settings
- the protocol used (HPGL, ...). Some devices (ex the circut #130 ) use encryption, if that's the case for this it must also be reversed before it can be used

DrDabadu, 2019-01-21:
Hi frmdstryr,
thanks for your kind and quick reply. I know that the device uses usb for connection to pc and wifi for connection to android & iOS devices.
Brother uses a special file format (FCM) but states, that the plotter can make use of svg files. It remains unclear whether the computer software converts the files prior to sending them to the plotter.

Seems like I have to keep searching for specs. When I find the time I try to get information from the Brother customer support. Maybe I'll buy one and see if it surprisingly works out of the box.

Thank you for the support

StafDehat, 2019-05-19:
I have a DC200 myself, and I'm running Mint 19.  I was able to get CanvasWorkspace running locally in WINE, which I can use to layout an SVG and send the results to my online DesignNCut account.  From there, I've been using the Android App on my Moto X4 (Project Fi) to initiate the cut.  I'd love to get InkCut working with it though, and I'm happy to run any debugging you may need to gather data.  Let me know if there's anything I can do to help.

justjanne, 2024-02-26:
If anyone is interested in adding support for it, I've written a free software implementation of brother's FCM format: https://github.com/justjanne/fcmlib

It should be relatively easy to at least add support for exporting tool paths as FCM now.