inkcut/inkcut #134: Cut off commando/choice

anITiot, 2019-02-09:
In the "weedlines" tab it should be nice to add a "cut off line" whit a selectable pressure just to cut off the vinyl. I have problems to cut straight lines with my scissors manually ;) or maybe lazy.
And a option to save a work with all the settings that have been used for a particular job.

frmdstryr, 2019-02-09:
Good idea. However, I'm not sure if the cutting knife is supposed to be doing sheet cutting. 

Have you tried just cutting a line with a high force? Does it work without screwing up the material? In my experience, any cut that goes through the backing usually messes up (folds, wrinkles, etc..) the material as it can lift part way through.

Some professional cutters (ex Roland)  have a sheet cut button with a separate cutting wheel that drops down to cut the sheet.  If your cutter has a command to do a sheet cut you can add it to the "After job" commands in the settings.

The "Jobs" pane should save and restore all the settings for a particular job, but it only saves per document and after it's cut.  Currently there's no option to just save as a new job (but that's also a good idea).  Can you add a separate issue for that?

anITiot, 2019-02-11:
Thanks for answer, will do a new issue for "save new job" :)

closed 2019-02-11 by anITiot