inkcut/inkcut #135: Fails to load on Ubuntu 18.04.2 LTS: No module named 'future'

akaziuna, 2019-02-10:
I'm using Ubuntu 18.04.2 LTS on a Parallels virtual machine, I've installed Inkcut according to the instructions: https://codelv.com/projects/inkcut/download/ but I can't get it to run. 

parallels@parallels-Parallels-Virtual-Platform:~$ inkcut
Traceback (most recent call last):
  File "/home/parallels/.local/bin/inkcut", line 11, in <module>
    sys.exit(main())
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/app.py", line 21, in main
    workbench.run()
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/core/workbench.py", line 127, in run
    plugin = self.get_plugin('inkcut.core')
  File "/home/parallels/.local/lib/python3.6/site-packages/enaml/workbench/workbench.py", line 145, in get_plugin
    plugin = manifest.factory()
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/core/manifest.enaml", line 60, in plugin_factory
    from inkcut.core.plugin import CorePlugin
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/core/plugin.py", line 18, in <module>
    from inkcut.core.api import Plugin, log
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/core/api.py", line 12, in <module>
    from .models import Model, Plugin, AreaBase
  File "/home/parallels/.local/lib/python3.6/site-packages/inkcut/core/models.py", line 17, in <module>
    from future.builtins import str
ModuleNotFoundError: No module named 'future'

frmdstryr, 2019-02-11:
Was there errors when running `pip3 install inkcut` ? 

Try `pip3 install future` (and do the same for other similar `No module named <something>` errors)

akaziuna, 2019-02-14:
No errors occurred when installing Inkcut. `pip3 install future` worked, thanks! And thank you for this software! 

I'm running on a virtual machine to get it working quickest, as there are no instructions yet for Mac and a bunch of dependencies. I just wanted to get my Roland Cam 1 CX-24 running again.

frmdstryr, 2019-02-14:
Great! Please reopen if needed.

closed 2019-02-14 by frmdstryr