inkcut/inkcut #136: Save a job for repeat use

anITiot, 2019-02-11:
Hi,  save a work as its looks when it is cut, the cutted size of the object, material, options. It will be easier to do the same thing again.
Or just save the commands that are send to the cutter/plotter.

Suggestion:
Save the job for the explicit model of cutter/plotter, material and settings that is used. No need  to make the job compatible for other models.

frmdstryr, 2019-02-14:
I added this, please give it at try. You can install it with `pip3 install git+https://github.com/codelv/inkcut.git`

There's a File->Save (or Ctrl+S) which will save the current settings to the Job list.

anITiot, 2019-02-17:
First, it seems to be working but I have swedishs setting of time and it couldn't save the job.
But the swap of x&y axle was great.
Then I tried to change to US time settings and that f*d up the whole installation.
Error box code:
An error prevented startup. This is either a bug or a required library is missing on your system. 

return pair.reader(owner, name)
  File "/home/steffe/.local/lib/python3.6/site-packages/enaml/core/standard_handlers.py", line 101, in __call__
    return call_func(func, (tr,), {}, scope)
  File "/home/steffe/.local/lib/python3.6/site-packages/inkcut/job/view.enaml", line 463, in <module>
    iterable << reversed(sorted(plugin.jobs, key=lambda j: j.info.created or datetime(1970, 1, 1)))
  File "/home/steffe/.local/lib/python3.6/site-packages/enaml/core/compiler_helpers.py", line 754, in wrapper
    return call_func(func, args, kwargs, scope)
  File "/home/steffe/.local/lib/python3.6/site-packages/inkcut/job/view.enaml", line 463, in <lambda>
    iterable << reversed(sorted(plugin.jobs, key=lambda j: j.info.created or datetime(1970, 1, 1)))
AttributeError: 'NoneType' object has no attribute 'created'

Please report this issue or request help at https://github.com/codelv/inkcut 
Error box code end:

Im running Ubuntu 18.04, the beaver edition. ;)
I tried to understand which python package to install but my python skills can only be better....

Very best regards and happy coding
//steffe
[inkcut.txt](https://github.com/codelv/inkcut/files/2873368/inkcut.txt)

frmdstryr, 2019-02-18:
Thanks for the log.  I'm not sure why, but one of the required fields not being saved & restored. It will now create the values if they're missing.  https://github.com/codelv/inkcut/commit/4d56e3effdd0731e1d15b283f2a3f71e7ef6e36b

anITiot, 2019-02-18:
An error report without a log is useless :)
Thanks for fixing the error on the material size etc.

But I like to name the job when I save it so I know what it is, like cutname_vinylsize_copies_weedlines_newsize_etc, extension name ".cut", now it feels like when I change the svg file in inkscape it will ruin the inkcut save. Well, it will be like a space-eater on the disk but not every job is worth to save.

And when I start inkcut I think it will be best if it starts "clean"/empty.

Great work anyway
//steffe

frmdstryr, 2019-02-18:
Inkcut only saves the configuration of each job, not the output.  I could make a way to attach a name to the job when saving it if that's what you're looking for?

If you want to save the output you can choose the "Save to file" in device setup then just rename those however you like (but they can't be re-opened into Inkcut).

anITiot, 2019-02-18:
I could make a way to attach a name to the job when saving it if that's what you're looking for?

Yes, that will be great :)

Tonight or tomorrow I will check if the integration with inkscape  do errors I see on my first installation, size and orientation errors.

Thanks
//steffe

anITiot, 2019-02-18:
Due to some user error (misunderstanding of program use) I had some setup errors, they are corrected now. It was orientation and size errors. Sorry.

Very best regards and thanks for a excellent program.
//steffe

frmdstryr, 2019-02-21:
Thanks please open a new issue if needed

closed 2019-02-21 by frmdstryr