inkcut/inkcut #138: Support for Silhouette Cameo (2?)

bianchifan, 2019-02-16:
I just did a fresh buy on ebay for a silhouette cameo 2 for I read some time in the past cameo/portrait would be supported.

I'm using Ubuntu 18.10 and tried two drivers, generic and Silhouette portrait (no cameo was listed).
But nothing happend! In Live view 100% were reached after 2/100 sec or similar.
So I switched to file print /tmp where inkcut12nn..gpgl files were generated.
ls -l pointed out a device /dev/usb/lp3,  cat ink... > /dev/usb/lp3 got the printer to life :)
I found issue #117 and tried "/dev/usb/lp3" as file as mentioned but it didn't work for me:
"/dev/usb/lp3/" is not a directory was the answer.

anITiot, 2019-02-20:
Did you try "Serial port" /dev/usb/lp3 or just "Printer"?

bianchifan, 2019-02-24:
At least I smashed my Ubuntu Installation and did a fresh Ubuntu 18.10.
Now all is runnin' fine at this point! :))
I can print directly and via temp-file plus cat.

closed 2019-02-24 by bianchifan