inkcut/inkcut #140: Swap X / Y axle

anITiot, 2019-02-22:


frmdstryr, 2019-02-24:
Yes I can add a swap x y option there

frmdstryr, 2019-05-21:
Added in https://github.com/codelv/inkcut/commit/cc5a155e293865f42df7ced0f128466119eace3f

closed 2019-05-21 by frmdstryr

lawrence-jeff, 2021-01-05:
@frmdstryr  I know this is a closed issue, but does this swap option not work using the control tab? My device (Vinyl Express/Redsail) has the axis swapped and no matter what setting I use the jog commands are still flipped. If I setup two identical devices one with the flip x/y enabled and one without and test using both the jog commands seem in both cases to send the same commands (i.e. it doesn't flip the values depending on profile).

Just getting started with the software so want to make sure its working as expected first.. maybe the control code doesn't respect this setting but actual printing does?