inkcut/inkcut #141: Using "Pen" failes at the end

bianchifan, 2019-02-24:
Plotter: Silhouette Cameo 2
Ubuntu 18.10
Ubuntu Studio 18.10

When plotting out is finished the tool remains on the last "down" position for about one second.
Using an ink-pen, i.e. fine liner, will soak the paper, the result is a **big point** at the end...

frmdstryr, 2019-02-24:
Which protocol is it using? 

Could you please save the output to a file (device settings) then paste the last few commands here?

bianchifan, 2019-02-25:
GPGL

tail of inkcut-1551086179.gpgl (c&p)
...D778,203D772,209D768,205D762,201D756,198D749,194M0,0

tail of logfile (couldn't be attached, "we can't process that file")
2019-02-25 10:16:20,131 | DEBUG   | inkcut | -> File | D778,203
2019-02-25 10:16:20,132 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,132 | DEBUG   | inkcut | -> File | D772,209
2019-02-25 10:16:20,133 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,133 | DEBUG   | inkcut | -> File | D768,205
2019-02-25 10:16:20,134 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,135 | DEBUG   | inkcut | -> File | D762,201
2019-02-25 10:16:20,135 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,136 | DEBUG   | inkcut | -> File | D756,198
2019-02-25 10:16:20,136 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,137 | DEBUG   | inkcut | -> File | D749,194
2019-02-25 10:16:20,137 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,139 | DEBUG   | inkcut | -> File | M0,0
2019-02-25 10:16:20,140 | DEBUG   | inkcut | Rate is :0
2019-02-25 10:16:20,140 | DEBUG   | inkcut | device | finish
2019-02-25 10:16:20,140 | DEBUG   | inkcut | device | disconnect
2019-02-25 10:16:20,140 | DEBUG   | inkcut | -- File | closed
2019-02-25 10:16:30,969 | ERROR   | root | Invalid alias: The name clear can't be aliased because it is another magic command.
2019-02-25 10:16:30,970 | ERROR   | root | Invalid alias: The name more can't be aliased because it is another magic command.
2019-02-25 10:16:30,970 | ERROR   | root | Invalid alias: The name less can't be aliased because it is another magic command.
2019-02-25 10:16:30,970 | ERROR   | root | Invalid alias: The name man can't be aliased because it is another magic command.