inkcut/inkcut #142: Help!

JackThomasStone, 2019-03-07:
An error prevented startup. This is either a bug or a required library is missing on your system. 

and only one is imported per session.

    Currently-imported Qt library:   'pyqt5'
    PyQt4 installed:                 False
    PyQt5 installed:                 False
    PySide >= 1.0.3 installed:       False
    PySide2 installed:               False
    Tried to load:                   ['pyqt5']
    
Loading from IPython < 4.0 failed with: No module named 'IPython.qt.console.rich_ipython_widget'

Please report this issue or request help at https://github.com/codelv/inkcut

frmdstryr, 2019-03-07:
Try `pip3 install pyqt5`

JackThomasStone, 2019-03-07:
> Try `pip3 install pyqt5`

Worked. Thank you! :-)

closed 2019-03-07 by frmdstryr