inkcut/inkcut #144: the ok button is to far down

ARKopp, 2019-03-17:
I cannot select ok to select my cutter and the protocol 

[image]

frmdstryr, 2019-03-22:
Do any of the other tabs have content that stretches the screen?

ARKopp, 2019-03-22:
Filter as well ... and i cannot make the taps smaller.
[image]

frmdstryr, 2019-03-22:
Can you post a screenshot of the device tab? I suspect it's not making them scroll like it does on other OS's.

ARKopp, 2019-03-22:
[image]

ARKopp, 2019-03-22:
i am on Mac OS High Sierra 10.13.6