inkcut/inkcut #146: problemens launching inkcut in ubuntu

ARKopp, 2019-03-30:
I get an inkcut-worspace startup error

An error prevented startup. This is either a bug or a required library is missing on your system. 

and only one is imported per session.

    Currently-imported Qt library:   'pyqt5'
    PyQt4 installed:                 False
    PyQt5 installed:                 False
    PySide >= 1.0.3 installed:       False
    PySide2 installed:               False
    Tried to load:                   ['pyqt5']
    
Loading from IPython < 4.0 failed with: No module named 'IPython.qt.console.rich_ipython_widget'

Please report this issue or request help at https://github.com/codelv/inkcut

ARKopp, 2019-03-30:
'''
sudo apt-get install python-qt4

frmdstryr, 2019-03-31:
Are you using python 3 or python 2? On python 2 you may need to run `pip install IPython`

frmdstryr, 2019-05-13:
Try `sudo apt install python3-pyqt5.qtsvg` Please reopen if needed.

closed 2019-05-13 by frmdstryr