inkcut/inkcut #148: Add i18n support

aaujon, 2019-04-05:
Hi,

It would be nice to have i18n support, I can give you a hand to dev this feature.
What library would you prefer to use : gettext, python-i18n, .. ?

frmdstryr, 2019-04-05:
I'd prefer using Qt's translation https://doc.qt.io/qt-5/internationalization.html

aaujon, 2019-04-08:
I started adding this feature here : https://github.com/aaujon/inkcut/commit/9a615a79421d21daa5cce405f1f1dadfeeedd12a does is seem fine to you ? If so I will push a PR when it's done.
Locale is auto-detected from system locale, which isn't great, I probably need to add a setting somewhere

frmdstryr, 2019-04-08:
Thanks for adding this!

I'd prefer using "config" or "device" instead of "transports" in `inkcut/device/view.enaml`. 

But it looks good to me. :shipit:

closed 2019-04-28 by aaujon