inkcut/inkcut #153: InkCut crashes on load in Windows 10

harbnger, 2019-04-22:
InkCut 2.0.7, loaded via "python -m pip install inkcut" Python 2.7.16 (x64)
Error message in popup shown here: [https://imgur.com/a/4RsKr4o](url)

Text from administrative console (where it was run from):
  File "c:\python27\lib\site-packages\inkcut\core\plugin.py", line 38, in start_default_workspace
    ui.select_workspace('inkcut.workspace')
  File "c:\python27\lib\site-packages\enaml\workbench\ui\ui_plugin.py", line 158, in select_workspace
    self._model.workspace = new_workspace
  File "c:\python27\lib\site-packages\enaml\core\standard_tracer.py", line 60, in __call__
    engine.update(owner, self.name)
  File "c:\python27\lib\site-packages\enaml\core\expression_engine.py", line 248, in update
    setattr(owner, name, pair.reader(owner, name))
  File "c:\python27\lib\site-packages\enaml\core\include.py", line 85, in _observe_objects
    self.parent.insert_children(self, newvalue)
  File "c:\python27\lib\site-packages\enaml\core\object.py", line 228, in insert_children
    child_added(child)
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 182, in child_added
    super(ToolkitObject, self).child_added(child)
  File "c:\python27\lib\site-packages\enaml\styling.py", line 343, in child_added
    super(Stylable, self).child_added(child)
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 141, in child_added
    child.initialize()
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 160, in initialize
    super(ToolkitObject, self).initialize()
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 118, in initialize
    child.initialize()
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 160, in initialize
    super(ToolkitObject, self).initialize()
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 118, in initialize
    child.initialize()
  File "c:\python27\lib\site-packages\enaml\core\include.py", line 42, in initialize
    obj.initialize()
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 160, in initialize
    super(ToolkitObject, self).initialize()
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 120, in initialize
    self.initialized()
  File "c:\python27\lib\site-packages\enaml\core\declarative_meta.py", line 68, in declarative_change_handler
    engine.write(owner, change['name'], change)
  File "c:\python27\lib\site-packages\enaml\core\expression_engine.py", line 217, in write
    pair.writer(owner, name, change)
  File "c:\python27\lib\site-packages\enaml\core\standard_handlers.py", line 82, in __call__
    call_func(func, (), {}, scope)
  File "c:\python27\lib\site-packages\inkcut\monitor\view.enaml", line 34, in f
    if connection:
  File "c:\python27\lib\site-packages\enaml\core\declarative_meta.py", line 43, in __call__
    value = engine.read(owner, name)
  File "c:\python27\lib\site-packages\enaml\core\expression_engine.py", line 186, in read
    return pair.reader(owner, name)
  File "c:\python27\lib\site-packages\enaml\core\standard_handlers.py", line 101, in __call__
    return call_func(func, (tr,), {}, scope)
  File "c:\python27\lib\site-packages\inkcut\monitor\view.enaml", line 29, in <module>
    attr connection << device.connection if device else None
  File "c:\python27\lib\site-packages\inkcut\device\plugin.py", line 402, in _default_connection
    return declaration.factory(driver, declaration, protocol)
  File "c:\python27\lib\site-packages\inkcut\device\transports\printer\manifest.enaml", line 15, in transport_factory
    from .plugin import PrinterTransport, Win32PrinterConfig, CupsPrinterConfig
  File "c:\python27\lib\site-packages\inkcut\device\transports\printer\plugin.py", line 153, in <module>
    class LPRPrinterConnection(PrinterConnection):
  File "c:\python27\lib\site-packages\inkcut\device\transports\printer\plugin.py", line 155, in LPRPrinterConnection
    _protocol = Instance(LPRProtocol)
  File "c:\python27\lib\site-packages\atom\instance.py", line 54, in __init__
    self.set_validate_mode(Validate.Instance, kind)
TypeError: Expected object of type `type or tuple of types`. Got object of type `classobj` instead.

2019-04-22 01:47:31,464 | CRITICAL | inkcut | An error prevented startup. This is either a bug or a required library is missing on your system.

return declaration.factory(driver, declaration, protocol)
  File "c:\python27\lib\site-packages\inkcut\device\transports\printer\manifest.enaml", line 15, in transport_factory
    from .plugin import PrinterTransport, Win32PrinterConfig, CupsPrinterConfig
  File "c:\python27\lib\site-packages\inkcut\device\transports\printer\plugin.py", line 153, in <module>
[... 102 lines cut ...]
    super(ToolkitObject, self).destroy()
  File "c:\python27\lib\site-packages\enaml\styling.py", line 298, in destroy
    super(Stylable, self).destroy()
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 129, in destroy
    super(Declarative, self).destroy()
  File "c:\python27\lib\site-packages\enaml\core\object.py", line 114, in destroy
    child.destroy()
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 169, in destroy
    super(ToolkitObject, self).destroy()
  File "c:\python27\lib\site-packages\enaml\styling.py", line 298, in destroy
    super(Stylable, self).destroy()
  File "c:\python27\lib\site-packages\enaml\core\declarative.py", line 129, in destroy
    super(Declarative, self).destroy()
  File "c:\python27\lib\site-packages\enaml\core\object.py", line 114, in destroy
    child.destroy()
  File "c:\python27\lib\site-packages\enaml\widgets\toolkit_object.py", line 172, in destroy
    self.proxy.destroy()
  File "c:\python27\lib\site-packages\enaml\qt\qt_ipython_console.py", line 83, in destroy
    self._teardown_kernel()
  File "c:\python27\lib\site-packages\enaml\qt\qt_ipython_console.py", line 113, in _teardown_kernel
    ipy_widget.kernel_client.stop_channels()
AttributeError: 'NoneType' object has no attribute 'kernel_client'

Windows Version [Version 10.0.17134.706] (x64).  
Machine's ports are all USB Serial interfaces. System is Core i7-8750H, 32GB RAM, installed on /dev/nvme1 (512 GB)

frmdstryr, 2019-04-22:
Thanks for reporting this, the change here https://github.com/codelv/inkcut/commit/306c027bcbe02490179f05c4ea192737e939810a should fix that error.

You can either make the same change or install the latest from github by running:

```
pip install https://github.com/codelv/inkcut.git
```

harbnger, 2019-05-02:
I apologize for a delayed response, I've had a busy week. 
Here's the result when I ran that command:

DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting https://github.com/codelv/inkcut.git
  Downloading https://github.com/codelv/inkcut.git
     / 122kB 939kB/s
  Cannot unpack file %User%\appdata\local\temp\pip-unpack-j29xbi\inkcut.git (downloaded from %User%\appdata\local\temp\pip-req-build-mtohfa, content-type: text/html; charset=utf-8); cannot detect archive format
Cannot determine archive format of %User%\appdata\local\temp\pip-req-build-mtohfa

frmdstryr, 2019-05-02:
Oops, sorry it should be

```
pip install git+https://github.com/codelv/inkcut.git
```

closed 2020-02-21 by frmdstryr