inkcut/inkcut #155: Inkscape plugin Mac

AntonioAmico, 2019-04-26:
Hi, I am getting the following error when I'm trying to lunch "Cut selection" from Inkscape:

Traceback (most recent call last):
 File "inkcut_cut.py", line 73, in <module>
  effect.affect()
 File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 283, in affect
  self.effect()
 File "inkcut_cut.py", line 68, in effect
  p.stdin.write(inkex.etree.tostring(document))
IOError: [Errno 32] Broken pipe

the log file say:
2019-04-26 08:53:30,279 | ERROR   | inkcut | Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/inkcut/app.py", line 73, in main
    from inkcut.core.workbench import InkcutWorkbench
  File "/usr/local/lib/python3.7/site-packages/inkcut/core/workbench.py", line 22, in <module>
    from inkcut.core.utils import log
  File "/usr/local/lib/python3.7/site-packages/inkcut/core/utils.py", line 21, in <module>
    from .svg import QtSvgDoc
  File "/usr/local/lib/python3.7/site-packages/inkcut/core/svg.py", line 20, in <module>
    from lxml import etree
ImportError: dlopen(/Applications/Inkscape.app/Contents/Resources/lib/python2.7/site-packages/lxml/etree.so, 2): Symbol not found: _PyBaseString_Type
  Referenced from: /Applications/Inkscape.app/Contents/Resources/lib/python2.7/site-packages/lxml/etree.so
  Expected in: flat namespace
 in /Applications/Inkscape.app/Contents/Resources/lib/python2.7/site-packages/lxml/etree.so

Can you help me?
[image]

frmdstryr, 2019-04-26:
Does `inkcut` start if you launch it from a terminal window?

AntonioAmico, 2019-04-26:
Yes, it starts, I can plot one time, then inkcut stops working and the Mac need to reboot

frmdstryr, 2019-04-26:
Thanks. Can you provide some more information...

What device are you using? How is it connected?

When Inkcut "stops working",  what does that mean? It locks up? Exits and doesn't start? Gives an error? etc...

Can you attach logs "Help -> Logs"?

frmdstryr, 2019-05-27:
Please reopen if needed.

closed 2019-05-27 by frmdstryr