inkcut/inkcut #159: Problem starting Inkcut on Ubuntu 19.04

badunc, 2019-04-28:
See attached log, please.
[inkcut.txt](https://github.com/codelv/inkcut/files/3124474/inkcut.txt)

frmdstryr, 2019-04-28:
`qtpy.PythonQtError: No Qt bindings could be found`

Did you run `pip3 install PyQt5` ? 

If you did that, try making sure qtconsole is installed `pip3 install qtconsole`

badunc, 2019-04-28:
Both commands returned "Requirement already satisfied".

Requirement already satisfied: PyQt5 in /usr/lib/python3/dist-packages (5.12.1)
Requirement already satisfied: qtconsole in ./.local/lib/python3.7/site-packages (4.4.3)
Requirement already satisfied: ipython-genutils in ./.local/lib/python3.7/site-packages (from qtconsole) (0.2.0)
Requirement already satisfied: ipykernel>=4.1 in ./.local/lib/python3.7/site-packages (from qtconsole) (5.1.0)
Requirement already satisfied: jupyter-core in ./.local/lib/python3.7/site-packages (from qtconsole) (4.4.0)
Requirement already satisfied: jupyter-client>=4.1 in ./.local/lib/python3.7/site-packages (from qtconsole) (5.2.4)
Requirement already satisfied: pygments in ./.local/lib/python3.7/site-packages (from qtconsole) (2.3.1)
Requirement already satisfied: traitlets in ./.local/lib/python3.7/site-packages (from qtconsole) (4.3.2)
Requirement already satisfied: tornado>=4.2 in ./.local/lib/python3.7/site-packages (from ipykernel>=4.1->qtconsole) (6.0.2)
Requirement already satisfied: ipython>=5.0.0 in ./.local/lib/python3.7/site-packages (from ipykernel>=4.1->qtconsole) (7.5.0)
Requirement already satisfied: pyzmq>=13 in ./.local/lib/python3.7/site-packages (from jupyter-client>=4.1->qtconsole) (18.0.1)
Requirement already satisfied: python-dateutil>=2.1 in /usr/lib/python3/dist-packages (from jupyter-client>=4.1->qtconsole) (2.7.3)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from traitlets->qtconsole) (1.12.0)
Requirement already satisfied: decorator in ./.local/lib/python3.7/site-packages (from traitlets->qtconsole) (4.4.0)
Requirement already satisfied: jedi>=0.10 in ./.local/lib/python3.7/site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (0.13.3)
Requirement already satisfied: backcall in ./.local/lib/python3.7/site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (0.1.0)
Requirement already satisfied: prompt-toolkit<2.1.0,>=2.0.0 in ./.local/lib/python3.7/site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (2.0.9)
Requirement already satisfied: setuptools>=18.5 in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (40.8.0)
Requirement already satisfied: pickleshare in ./.local/lib/python3.7/site-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (0.7.5)
Requirement already satisfied: pexpect; sys_platform != "win32" in /usr/lib/python3/dist-packages (from ipython>=5.0.0->ipykernel>=4.1->qtconsole) (4.6.0)
Requirement already satisfied: parso>=0.3.0 in ./.local/lib/python3.7/site-packages (from jedi>=0.10->ipython>=5.0.0->ipykernel>=4.1->qtconsole) (0.4.0)
Requirement already satisfied: wcwidth in ./.local/lib/python3.7/site-packages (from prompt-toolkit<2.1.0,>=2.0.0->ipython>=5.0.0->ipykernel>=4.1->qtconsole) (0.1.7)

frmdstryr, 2019-04-28:
If you run  `ipython qtconsole` does that also give an error?

badunc, 2019-04-28:
Yes, it does:
Traceback (most recent call last):
  File "/home/bduncan/.local/bin/ipython", line 10, in <module>
    sys.exit(start_ipython())
  File "/home/bduncan/.local/lib/python2.7/site-packages/IPython/__init__.py", line 119, in start_ipython
    return launch_new_instance(argv=argv, **kwargs)
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 657, in launch_instance
    app.initialize(argv)
  File "</home/bduncan/.local/lib/python2.7/site-packages/decorator.pyc:decorator-gen-110>", line 2, in initialize
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/home/bduncan/.local/lib/python2.7/site-packages/IPython/terminal/ipapp.py", line 307, in initialize
    super(TerminalIPythonApp, self).initialize(argv)
  File "</home/bduncan/.local/lib/python2.7/site-packages/decorator.pyc:decorator-gen-7>", line 2, in initialize
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/home/bduncan/.local/lib/python2.7/site-packages/IPython/core/application.py", line 455, in initialize
    self.parse_command_line(argv)
  File "/home/bduncan/.local/lib/python2.7/site-packages/IPython/terminal/ipapp.py", line 302, in parse_command_line
    return super(TerminalIPythonApp, self).parse_command_line(argv)
  File "</home/bduncan/.local/lib/python2.7/site-packages/decorator.pyc:decorator-gen-4>", line 2, in parse_command_line
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 514, in parse_command_line
    return self.initialize_subcommand(subc, subargv)
  File "/home/bduncan/.local/lib/python2.7/site-packages/IPython/core/application.py", line 244, in initialize_subcommand
    return super(BaseIPythonApplication, self).initialize_subcommand(subc, argv)
  File "</home/bduncan/.local/lib/python2.7/site-packages/decorator.pyc:decorator-gen-3>", line 2, in initialize_subcommand
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/home/bduncan/.local/lib/python2.7/site-packages/traitlets/config/application.py", line 445, in initialize_subcommand
    subapp = import_item(subapp)
  File "/home/bduncan/.local/lib/python2.7/site-packages/ipython_genutils/importstring.py", line 31, in import_item
    module = __import__(package, fromlist=[obj])
ImportError: No module named qtconsole.qtconsoleapp

The command recommended that I run jupyter qtconsole instead so here's the output from that:

Traceback (most recent call last):
  File "/home/bduncan/.local/bin/jupyter-qtconsole", line 6, in <module>
    from qtconsole.qtconsoleapp import main
  File "/home/bduncan/.local/lib/python3.7/site-packages/qtconsole/qtconsoleapp.py", line 60, in <module>
    from qtconsole.qt import QtCore, QtGui
  File "/home/bduncan/.local/lib/python3.7/site-packages/qtconsole/qt.py", line 23, in <module>
    QtCore, QtGui, QtSvg, QT_API = load_qt(api_opts)
  File "/home/bduncan/.local/lib/python3.7/site-packages/qtconsole/qt_loaders.py", line 387, in load_qt
    api_options))
ImportError: 
    Could not load requested Qt binding. Please ensure that
    PyQt4 >= 4.7, PyQt5, PySide >= 1.0.3 or PySide2 is available,
    and only one is imported per session.

    Currently-imported Qt library:   None
    PyQt4 installed:                 False
    PyQt5 installed:                 False
    PySide >= 1.0.3 installed:       False
    PySide2 installed:               False
    Tried to load:                   ['pyqt5', 'pyside2', 'pyside', 'pyqt']

frmdstryr, 2019-04-28:
Ok thanks. It looks like it's mixing system versions of python packages and "user" versions (I really dislike how python handles 2.7 / 3.x and system vs user packages). 

Try running `pip3 install --user PyQt5` and see if that works (the user flag is important).

badunc, 2019-04-28:
No love there. :(

Requirement already satisfied: PyQt5 in /usr/lib/python3/dist-packages (5.12.1)

frmdstryr, 2019-04-29:
Shot in the dark here... run `sudo apt install python3-pyqt5.qtsvg` and see if that works

badunc, 2019-04-29:
Eureka!  Did the install and then Inkcut fired right up.  Thanks for the help!!!!

frmdstryr, 2019-04-29:
Great!

closed 2019-04-29 by frmdstryr