inkcut/inkcut #161: Cutting with a MH 721 on Mac? Not sure next steps

drewrwilson, 2019-05-10:
Hey! I love your project! I'm so impressed! I like that it's standalone software and also a inkscape plugin. 

I'm trying to get it working with my new vinyl cutter model MH721 (US Cutter) on MacOS via brew. I'm not sure what the next steps are. I installed InkCut standalone and the plugin in InkScape, but I'm not sure what the next steps are. Is there a driver I'm supposed to be installing? Any advice?

[image]

frmdstryr, 2019-05-10:
1. Go to Device -> Setup
2. Select "Inkcut generic driver"
3. Under the Connections tab choose type "Serial Port", set the port to the first one
4. Set protocol to HPGL
5. Press OK to close the dialog
6. Then Choose "Device - > Send to device"

drewrwilson, 2019-05-10:
Thanks for the lightning-fast response! It appears as though I don't have that dropdown menu.

[image]

frmdstryr, 2019-05-10:
Apparently OSX moves it to the "Python" menu and renames it to "Preferences"

drewrwilson, 2019-05-13:
I ordered a serial to USB cable and I'm going to try this again today

drewrwilson, 2019-05-13:
Update: Still didn't get it working. 

I tried it with the "Inkcut generic driver" and with the "US Cutter Refine MH-721".

I've tried with the USB AB cable that came with the vinyl cutter and also with a serial-to-USB cable that I ordered on amazon. This one: [https://smile.amazon.com/gp/product/B00IDSM6BW/](https://smile.amazon.com/gp/product/B00IDSM6BW/)

[image]

I read somewhere that this adapter might work better for MacOS: [https://smile.amazon.com/gp/product/B0000VYJRY](https://smile.amazon.com/gp/product/B0000VYJRY)

frmdstryr, 2019-05-13:
Any usb - serial cable should work. Most cutters have them builtin if there's a usb port.

drewrwilson, 2019-05-13:
Yeah, makes sense. These are the ports I have

[image]

drewrwilson, 2019-05-16:
wasn't able to get it working on MacOS, talked to the support people at US Cutter for a while and they said MacOS doesn't work reliably with this cutter. I got the cutter working on a Windows machine with the software that ships with the cutter, but I'll try to get inkcut working on this windows machine instead. I was able to install the dependencies, but I got an error from pip while installing inkcut on windows.

drewrwilson, 2019-05-16:
You can close this issue because I'm giving up getting it working on MacOS. Thank you for your responses!

raboof, 2019-05-16:
> talked to the support people at US Cutter for a while and they said MacOS doesn't work reliably with this cutter

That sounds kind of weird - I can imagine the US Cutter software software not working reliably on MacOS, but I'd expect the cutter itself to work fine...

> I'll try to get inkcut working on this windows machine instead

That sounds good! And if we can figure out how to make it work on Windows, we can revisit whether we can make the same also hapen on Mac

> I got an error from pip while installing inkcut on windows.

Feel free to open another ticket to discuss this ;)

closed 2019-05-27 by frmdstryr

flandish, 2020-09-20:
**For those still looking for an answer:** The cable has to have an **FTDI** chip in it. 

This Star Tech USB to Serial cable worked for me: https://www.amazon.co.uk/dp/B004ZMYTYC/ref=pe_385721_51767431_TE_dp_1

Took a lot troubleshooting to figure it out.