inkcut/inkcut #162: First attempt installing inkcut standalone on a raspberry pi3 raspbian 9.8 stretch

LennethW, 2019-05-12:
Clean image freshly etched. I tried the following:

- [x] sudo apt-get install python3-pip python3-pyqt5 libcups2-dev
- [x] pip3 install inkcut

Had an error on lxml, which mentioned to check if libxml2 was installed.
i then did

- [x] sudo apt-get install libxml2 libxml2-dev

after another pip3 install inkcut, i had a nice exit saying everything was okay and installed.
tried inkcut in terminal, had a "no command found" error.

searched and located binaries with pcmanfm in /home/pi/.local/bin
ran it from there , it throws the errors in attached files 

[inkcut.txt](https://github.com/codelv/inkcut/files/3169948/inkcut.txt)
[crash.txt](https://github.com/codelv/inkcut/files/3169947/crash.txt)

tried also 
- [x] pip3 install git+https://github.com/codelv/inkcut.git

and i get the same results.

frmdstryr, 2019-05-12:
I think the docs need updated. Try the command here https://github.com/codelv/inkcut/issues/159#issuecomment-487542680

LennethW, 2019-05-12:
Thank you very very very much.
I totally missed that, now my inkcut build is running finally! You saved my day

closed 2019-05-12 by LennethW