inkcut/inkcut #163: crashes at device setup

lgroc, 2019-05-13:
i'm trying to setup a Creation pcut cs-1200 with serial port but it crashes instantly at setup or refreshing port.

OS: windows 7 x32

log: [inkcut.txt](https://github.com/codelv/inkcut/files/3171097/inkcut.txt)

frmdstryr, 2019-05-13:
Is there a `crash.txt` in the log folder?

lgroc, 2019-05-13:
no, only inkcut.txt

frmdstryr, 2019-05-14:
Unfortunately I can't see anything in the log... Does it output any error message that is not reported in the log?

You can try to update to the latest version with  `pip3 install git+https://github.com/codelv/inkcut.git` there was one change related to printers that may have been the issue.

Bargud, 2020-02-21:
Good day,

I would like to ask if you already solve this problem because I have same issue. Once I opened device settings incut crash without any crash log, only incut.txt
If I do same thing with plotter disconnected from computer there is no issue.

I'm using version - v2.2.1

Thank you

closed 2020-02-22 by frmdstryr