inkcut/inkcut #165: ERROR

MaurinElectroTextile, 2019-05-27:
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch
Inkscape: 0.92

> Traceback (most recent call last):
>   File "inkcut_cut.py", line 73, in <module>
>     effect.affect()
>   File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
>     self.effect()
>   File "inkcut_cut.py", line 67, in effect
>     close_fds=sys.platform != "win32")
>   File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
>     errread, errwrite)
>   File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
>     raise child_exception
> OSError: [Errno 2] No such file or directory

Any idea ?

frmdstryr, 2019-05-27:
Inkcut must be installed in addition to the plugin for inkscape. See https://codelv.com/projects/inkcut/docs/installing

TrendMend, 2019-05-30:
To anyone having the same issue: Use sudo! When you don't use sudo, pip installs it to your home folder, so it will only be able to be used by you, the plugin doesn't know this and expects it to be systemwide. Just do the commands you need in the docs linked above, with sudo or root.