inkcut/inkcut #166: Wireless Control

TrendMend, 2019-05-30:
Would there be any possibility in adding a wireless control functionality? As in, hooking up a RPI or other small device to your plotter/cutter and then it could communicate with the master (your computer) for fully wireless functionality. You could add a randomized pin to keep people from connecting to it on your network, for safety purposes. This would be great for people who have their plotter/cutter not close to their computer, or would like to use a laptop without having to plug it in close to the plotter/cutter. Kind've like octoprint minus the online interface.

frmdstryr, 2019-05-30:
It'd be possible, it depends how much interest there would be for this.  Adding an rpc interface would be fine but if you're using a pi you could also just well use vnc.

I was thinking of making a simple wifi/usb interface https://inkcut.org/t/poll-interested-in-cutting-over-wifi-bluetooth/11/ but it seems to not be of interest to many people (or too early).  I've only had two requests for something like this so far.

TrendMend, 2019-05-30:
Thanks for the information, I was just saying the pi as an example, even something as overpowered as a NUC would do, but I love the idea of a interface for the sole purpose of inkcut, this would cut costs down by a good bit. If you ever want to do this, I am open to helping as I have some experience in Eagle CAD and electronics engineering (the last part is a W.I.P for me haha).

TrendMend, 2019-06-28:
So after some stackexchange searching, I found this: https://www.raspberrypi.org/forums/viewtopic.php?t=152391 I believe with the rpi zero w being around 10 dollars, this would be the way to go. The only next thing would be to make a ready image so it is just flash and go (minus some wifi configuration in a file)

TrendMend, 2019-06-28:
Oh, and the best part is: It is free for one device! that means that you wouldn't have to buy the license, this allows you to be able to use one usb device on the server, and an unlimited amount of clients for free. This is something that would be good for a makerspace where you can download the client application on your device and use it wirelessly.

Software name: virtualhere (actually needed for steam link, for a price)

jed-frey, 2020-01-26:
For my Silhouette Portrait, I've gotten this to work by sharing the printer via CUPS, adding it to my other machines and then just printing. CUPS handles the spooling and sending it to the device attached to the cutter.