inkcut/inkcut #168: Can't see USB port on linux (Lubuntu)

drewrwilson, 2019-06-06:
Hi! I'm trying to get my US-Cutter MH 721-MK2 working with an Acer netbook running Lubuntu (version: 18.04.2 bionic) and I can't get inkcut to recognize my usb ports. There is nothing in the drop down. I tested the usb port by using a USB drive in the same port and it worked. And then I checked `lsusb` and it appears that the system is seeing something. Screenshot below of `lsusb` output and also the dropdown in inkcut with no options. Any advice?

[image]

I'm very excited to use inkcut as my vinyl cutter software! Thank you for all your work on this project! :)

frmdstryr, 2019-06-06:
Honestly it sounds like a problem with the cutter. My guess is they probably used a knock off USB/serial chip that was bricked or the control board is just dead. Not uncommon for these cheap no-name brands.

raboof, 2019-06-06:
Is this the `7447:5419` or the `0402:7675`? A quick google suggests `0402:7675` looks like a webcam?

drewrwilson, 2019-06-06:
thanks for the quick replies!

@frmdstryr that's a good idea, but I have gotten the cutter working so I think the usb board works. I got it working on a Windows machine that I borrowed to test out the machine. 

@raboof I think that it's `7447:5419` because when I remove it and do a `lsbusb` that one goes away and then when I plug it in, `7447:5419` appears in the list. 

Some other info:
When I try to use the "Raw device descriptor" dropdown and put in `/dev/usb/lp0` I get an error that says permission denied. Screenshot below and I am including the log file in case that's helpful!

Trying to use "Raw device descriptor":
[image]

Error after trying to send to device:
[image]

Log file:
(i started with a fresh log file for this test, just to make it easier to read. So this log file is all from one session.)
[inkcut.txt](https://github.com/inkcut/inkcut/files/3262105/inkcut.txt)

Any thoughts? Maybe any permissions things I should try?

raboof, 2019-06-06:
when you run `sudo dmesg -w` and insert the USB device, what logging gets added there?

drewrwilson, 2019-06-06:
Ok, I ran `sudo dmesg -w`. It had a lot of misc output. I had ran it with my cutter unplugged from the usb port. After all the initial messages finished, I then plugged in my usb cable plugged into the cutter and this was the output:

```
[ 8880.948210] usb 2-1: new full-speed USB device number 14 using uhci_hcd
[ 8881.122270] usb 2-1: New USB device found, idVendor=7447, idProduct=5419, bcdDevice=12.08
[ 8881.122281] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 8881.129717] usblp 2-1:1.0: usblp0: USB Unidirectional printer dev 14 if 0 alt 0 proto 1 vid 0x7447 pid 0x5419
```

raboof, 2019-06-06:
ok cool. what is the output for `groups` and for `ls -alFh /dev/usb/lp0` when the machine is plugged in?

drewrwilson, 2019-06-06:
Ok here's the output for `groups`:

```
roman adm cdrom sudo dip plugdev lpadmin sambashare
```

And here's the output for `ls -alFh /dev/usb/lp0`:

```
crw-rw---- 1 root lp 180, 0 Jun  6 11:04 /dev/usb/lp0
```

raboof, 2019-06-06:
OK, so you can't access it because you're not in the `lp` group.

drewrwilson, 2019-06-06:
Ok, makes sense. I'm always a little confused by linux permissions. I'm googling around now. Do you know how to change it? 

I could try that and then see if inkcut can access the cutter

raboof, 2019-06-06:
'groupadd' or 'addgroup' should do the trick. you might have to log out and log in again to make it take effect.

drewrwilson, 2019-06-06:
GOOD NEWS! It worked after I added my default user to the `lp` group and restarted

[image]

So my cutter still doesn't show up in the dropdown list of possible serial connections, but if I type `/dev/usb/lp0` in the "Raw device descriptor" I can send a job to the cutter and it cuts! Hooray! 

For this test one I selected open mode from the dropdown to be `r+b`. And it worked. But actually I don't know what that means. There are also other options: `wb`, `r+`, and `w`. Any advice about which I should use or what the difference is? 

In any case, thank you thank you thank you!

drewrwilson, 2019-06-06:
thank you both for all your work on this open source project! i just chipped in $10 as a thank you via this page https://codelv.com/donate/

frmdstryr, 2019-06-06:
Good to hear. Thank you for the donation and to @raboof for the debugging help!

> So my cutter still doesn't show up in the dropdown list of possible serial connections

The cutter is showing up as a usb-parallel port, not serial, which is why it's not in the Serial port list.  Perhaps an option for parallel ports should be added.  You could also add it as a system printer and chose it from the printer list.

> Any advice about which I should use or what the difference is?

These are different file open modes python uses. Use `r+b` as it's the most flexible (open for read & write in binary mode). See https://docs.python.org/3/tutorial/inputoutput.html#reading-and-writing-files for more info if interested.

drewrwilson, 2019-06-06:
Thanks for explaining the different open modes! 

Closing this issue now

closed 2019-06-06 by drewrwilson

clayhands, 2020-06-28:
@raboof & @frmdstryr just wanted to say a big thank you for your guidance and info posted above^^  Found this issue while googling, and was able to get my Roland CAMM-1 GS-24 working with Inkcut on `/dev/usb/lp0` successfully following a similar journey as the original poster. Excited to learn what I can do with this, and ditch a VirtualBox Windows solely for running the cutter.