inkcut/inkcut #171: 2.0.7 and 2.1dev crash at startup

methiDD, 2019-06-13:
hallo,
I'm not able to start 2.0.7 and the dev version - it crashes at startup.
System: macOS 10.13 with Python 3

here the terminal output starting 2.0.7:
```
(...)
2019-06-13 22:55:48,739 | DEBUG   | inkcut | Starting plugin 'inkcut.ui'
2019-06-13 22:55:52,226 | DEBUG   | asyncio | Using selector: KqueueSelector
2019-06-13 22:55:52,267 | DEBUG   | asyncio | Using selector: KqueueSelector
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/qt/qt_abstract_item_view.py", line 73, in data
    h, v = d.text_alignment
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/widgets/abstract_item.py", line 142, in _update_proxy
    self.proxy.data_changed(change)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/qt/qt_table_view.py", line 196, in data_changed
    self.view.model.dataChanged.emit(self.index, self.index)
TypeError: QAbstractItemModel.dataChanged(): argument 1 has unexpected type 'NoneType'
Fatal Python error: Aborted

Thread 0x000070000a2b7000 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/selectors.py", line 558 in select
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 1739 in _run_once
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 539 in run_forever
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tornado/platform/asyncio.py", line 132 in start
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 865 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x0000700009d31000 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 296 in wait
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 552 in wait
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/IPython/core/history.py", line 829 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/decorator.py:decorator-gen-24>", line 2 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0x00007fff9d927380 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/qt/qt_table_view.py", line 83 in set_horizontal_stretch
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/qt/qt_abstract_item_view.py", line 195 in init_widget
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enamlx/qt/qt_table_view.py", line 59 in init_widget
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_toolkit_object.py", line 76 in activate_top_down
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 226 in activate_top_down
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 210 in activate_proxy
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213 in activate_proxy
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213 in activate_proxy
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213 in activate_proxy
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213 in activate_proxy
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 185 in child_added
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/object.py", line 228 in insert_children
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/include.py", line 85 in _observe_objects
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/expression_engine.py", line 248 in update
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/standard_tracer.py", line 61 in __call__
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/workbench/ui/ui_plugin.py", line 158 in select_workspace
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/core/plugin.py", line 38 in start_default_workspace
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/q_deferred_caller.py", line 42 in customEvent
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_application.py", line 48 in start
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/workbench/ui/ui_plugin.py", line 105 in start_application
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/core/workbench.py", line 160 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/app.py", line 75 in main
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/inkcut", line 11 in <module>
Abort trap: 6
```

Do you need any other informations?

By the way, 2.0.6 is starting, but crashes when entering the preferences.

Thanks for your help and work! I really like Inkcut!

frmdstryr, 2019-06-14:
Thanks for providing the log, it looks like the version of enamlx is an older as the current version checks for this case [here](https://github.com/frmdstryr/enamlx/blob/master/enamlx/qt/qt_table_view.py#L197)

Try running `pip3 install enamlx==0.4.1`

methiDD, 2019-06-16:
Thanks for the advice. 0.3.0 was installed.

Now 2.0.7 starts again, but still crash when opening the preferences/ settings.
This is the log, after clicking the menu:
Hope you can help me again.

Thanks!

```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/q_deferred_caller.py", line 42, in customEvent
    event.callback(*event.args, **event.kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/workbench/ui/workbench_menus.enaml", line 24, in invoke_action
    core.invoke_command(item.command, item.parameters, item)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/workbench/core/core_plugin.py", line 78, in invoke_command
    return command.handler(event)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/device/manifest.enaml", line 27, in device_setup
    DeviceDialog(ui.window, plugin=plugin).exec_()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/enamldef_meta.py", line 79, in __call__
    instance.__init__(parent, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/object.py", line 100, in __init__
    self.set_parent(parent)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/core/object.py", line 160, in set_parent
    parent.child_added(self)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 185, in child_added
    child.activate_proxy()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
    child.activate_proxy()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
    child.activate_proxy()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 213, in activate_proxy
    child.activate_proxy()
  [Previous line repeated 6 more times]
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 210, in activate_proxy
    self.activate_top_down()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/widgets/toolkit_object.py", line 226, in activate_top_down
    self.proxy.activate_top_down()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_toolkit_object.py", line 76, in activate_top_down
    self.init_widget()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_object_combo.py", line 95, in init_widget
    self.refresh_items()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_object_combo.py", line 148, in refresh_items
    if item == selected:
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/serial/tools/list_ports_common.py", line 74, in __eq__
    return self.device == other.device
AttributeError: 'NoneType' object has no attribute 'device'
Fatal Python error: Aborted

Thread 0x000070000efe4000 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/selectors.py", line 558 in select
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 1739 in _run_once
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/asyncio/base_events.py", line 539 in run_forever
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/tornado/platform/asyncio.py", line 132 in start
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/ipykernel/iostream.py", line 78 in _thread_main
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 865 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x000070000ea5e000 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 296 in wait
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 552 in wait
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/IPython/core/history.py", line 829 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/IPython/core/history.py", line 58 in needs_sqlite
  File "</Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/decorator.py:decorator-gen-24>", line 2 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0x00007fff9d927380 (most recent call first):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/qt/qt_application.py", line 48 in start
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/enaml/workbench/ui/ui_plugin.py", line 105 in start_application
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/core/workbench.py", line 160 in run
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/inkcut/app.py", line 75 in main
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/inkcut", line 11 in <module>
Abort trap: 6

```

frmdstryr, 2019-06-17:
The fix for this was added in pyserial a year ago in https://github.com/pyserial/pyserial/issues/286

There still has not been a new release with it in yet.   You can try installing pyserial from the master branch.  I may just need to add a workaround for it.

methiDD, 2019-07-19:
Hey again, i got the commit for this fix of pyserial and it worked. Thanks.

closed 2019-07-19 by methiDD